steve-community/steve #114: Problem on database Migration

Soulweed, 2019-03-01:
`Failed to execute goal org.flywaydb:flyway-maven-plugin:5.1.4:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: 
[ERROR] Migration V0_9_7__update.sql failed
[ERROR] -----------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1349
[ERROR] Message    : View's SELECT contains a subquery in the FROM clause
[ERROR] Location   : /home/pi/workspace/steve/src/main/resources/db/migration/V0_9_7__update.sql (/home/pi/workspace/steve/src/main/resources/db/migration/V0_9_7__update.sql)
[ERROR] Line       : 5
[ERROR] Statement  : CREATE OR REPLACE VIEW ocpp_tag_activity AS
[ERROR]     SELECT
[ERROR]       ocpp_tag.*,
[ERROR]       COALESCE(tx_activity.active_transaction_count, 0) as 'active_transaction_count',
[ERROR]       CASE WHEN (active_transaction_count > 0) THEN TRUE ELSE FALSE END AS 'in_transaction'
[ERROR]     FROM ocpp_tag
[ERROR]     LEFT JOIN
[ERROR]     (SELECT id_tag, count(id_tag) as 'active_transaction_count'
[ERROR]       FROM transaction
[ERROR]       WHERE stop_timestamp IS NULL
[ERROR]       AND stop_value IS NULL
[ERROR]       GROUP BY id_tag) tx_activity
[ERROR]     ON ocpp_tag.id_tag = tx_activity.id_tag`

goekay, 2019-03-01:
select subqueries are not supported by mysql versions below 5.7.7. is this your case? that's why we suggest using at least mysql 5.7.7 in readme.

closed 2019-03-02 by goekay

Soulweed, 2019-03-07:
Thank you very much. in case of raspberryPi, I clone the releases of 3.2.0