steve-community/steve #115: Powermeter data empty

taston, 2019-03-04:
I have Chago Ensto charger. I have problems with power consumption usage.

In Charger I set meter values to:

[image]

But in database I see only "zero values":

[image]

Any suggestions to set it to works propertly?
Or maybe I can debug data received from charger?

goekay, 2019-03-04:
hi, we log the communication with the station (requests and responses) to the log file. i would advise to look for messages with the same timestamps whether the value is actually set in the message.

goekay, 2019-03-20:
closing this issue due to its inactivity

closed 2019-03-20 by goekay