steve-community/steve #116: KEBA P30 configuration

jhs-s, 2019-03-09:
How to configure a KEBA P30 X-Series to connect to steve?

Steve is running behing a reverse proxy (nginx) on a subdomain with default redirection to HTTPS and Port 443 redirecting internally. This works flawlessly for the webinterface.
The Keba-Box ist configured with the SOAP URL https://xzy.info:443/steve/services/CentralSystemService and says that the backend is reachable. The box is configured in steve and I even enabled automatic addition of unknown stations in the config file. The box internal port has been added to the firewall so that it should be reachable.

However, I cannot get any heartbeat and there is nothing in the log file showing up about a box trying to connect.

Is there any example configuration?

Thanks a lot

csamsel, 2019-03-10:
If nothing is in the log file, the station doesn't send any messages. I don't that model, so i cannot help you with its configuration.

But, although the reverse proxy configuration should work, i'd suggest starting with a simple configuration to rule out any errors there. I'd also check the communication with wireshark.

goekay, 2019-03-11:
> ... https://xzy.info:443/steve/services/CentralSystemService and says that the backend is reachable

let's assume that we do not trust this claim of the station. you can try to load `https://xzy.info:443/steve/services` using any browser. the page should list all four soap services that are up and running. if it is the case, you can be sure that steve and its soap services are reachable behind your reverse proxy. then it is pretty safe to conclude that the problem lies with the station.

jhs-s, 2019-03-25:
Sorry for the late reply, I finally had the time to setup everything for wireshark logging. Following happens:
Keba P30 Request: HEAD http://boxtest.dnshome.de/steve/services/CentralSystemService?wsdl HTTPS 1.0
Steve reply: 400 HTTP 1.1 Unknown version

The magic question is why the box is sending a http request including "HTTPS 1.0". There is a "secure connection" configuration parameter which is set to false, setting it to true actually changes the URL to https and the connection fails completely. 

@goekay: The services show up - but you are right that it is easier to debug without the reverse proxy.

Any clues? Thanks for your help!

csamsel, 2019-03-25:
> The magic question is why the box is sending a http request including "HTTPS 1.0".

broken implementation. That's very common for charging stations.

goekay, 2019-03-25:
i agree. the request ending with `HTTPS 1.0` does not seem to be valid [1]. our java server library jetty cannot recognize this string (because it's not one of the values in [2]), so you get an unknown version response back.

[1] https://developer.mozilla.org/en-US/docs/Web/HTTP/Basics_of_HTTP/Evolution_of_HTTP
[2] https://github.com/eclipse/jetty.project/blob/jetty-9.4.x/jetty-http/src/main/java/org/eclipse/jetty/http/HttpVersion.java#L31-L34

goekay, 2019-03-25:
i find this situation funny, because @micha17 added this model in [our compatibility list](https://github.com/RWTH-i5-IDSG/steve/wiki/Charging-Station-Compatibility). @micha17, can you chime in on this?

goRaspy, 2019-03-25:
As i know there is many hardware version for the same commercial name on
KEBA range.

jhs-s, 2019-03-26:
OK, I reported this to KEBA.
For reference: KeContact P30 x-Series SW: 1.8.2 (latest as of today)

I also checked the nginx log of the reverse proxy and not surprisingly it also sees that "HTTPS/1.0" requests and nginx answers with Code 400 before anything is forwarded to steve at all. As even nginx doesn't support this version code I am quite confident that it isn't used anywhere.

I will keep you updated here, I hope it is OK to keep this issue open even as it is not a bug in steve.

csamsel, 2019-03-26:
> I will keep you updated here, I hope it is OK to keep this issue open even as it is not a bug in steve.

Sure. 

I really wonder how bugs like this end up in products delivered to customers. 

a) It's like the OCPP implementation of the charging station has never been tested.
b) It appears to me that the developers implemented HTTP themselves instead of using a library (and failed miserably at it) ... i dont even want to know about its security.

micha17, 2019-03-27:
I installed Steve on the Rasperry step by step after the description here in the projectsite. The Keba socket uses the firmware 1.7.0. If there are no security issues, I generally wait for the experience of others with new firmware updates.
The computer is connected to the Keba socket via LAN. As protocol I use OCPP1.5 between Steve and Keba. I had no really problems by installing it. Fiddly was only to generated the keytool to activate HTTPS.

jhs-s, 2019-03-29:
After discussing the issue with the quite competent and fast KEBA-Support I finally got it working.

Pitfalls:
* The firmware doesn't accept letsencrypt by Default (will be fixed in 1.9)
* Some certificates (like mine) cannot be imported at all - KEBA sent me a patch for this (will be fixed in 1.9)
* The box actually stops trying to sent BootNotifications after 7 days when it cannot reach the server. It won't resume until the box is restarted, any configuration changes don't have an immediate effect.
* The http request shown above seems only to be used on the status page for a "ping" like feature which just ensures that the backend is available but has nothing to do with OCPP at all. Still unsure why they use HTTPS 1.0 there but it doesn't matter in general as for them a status code 400 seems to be a postive answer. Together with the fact that the box doesn't send request (see above) and as such this is the only communication that is happening makes this even more weird.

Anyway it works now and I can really recommend their techsupport!

closed 2019-03-29 by jhs-s