steve-community/steve #117: Integrate StEVe as a HomeAssistant / HASS.io plugin for automation

DrFrankReade, 2019-03-11:
Feature Request - A lot of good, forward progress would be made by bringing StEVe to this platform.

HomeAssistant  (aka HASS.io) is a fantastic program for self-hosted home (or whatever) automation with integrations for nearly everything one could imagine, plus open source and under very active development. Runs on a Raspberry Pi or anything else one could spin up as a server. Great interface, easily customized, need I go on?  Plugins run as servers on the same machine (or different) sometimes dockered and commands get passed back and forth. Since HASS integrates with so many other systems, utility providers, sensors, etc, the opportunity for automation and more external interface is immense. 

Adding easy OCPP to this very accessible platform would provide a great deal of utility for people with smart chargers that up until this point have been very dumb. 

Thank you for all of your efforts thus far. The work of your team is very important.

TheCrey, 2019-03-18:
Count me too, this is exactly what I have been looking for. 
I am using the server now since some days for two Amtron's with a ACU.
Would be nice to get en info if it is done, so I could let me send a message that charging is complete oder there was an issue and charging was stopped. 

I think a "simple" hook request could work as HA does have webhook component. 

Unfortunately i am not able to do something like this.

goekay, 2019-03-18:
i appreciate the interest, but this seems like a _huge_ undertaking, which i don't really have time for in the near future. having said that, i welcome any community effort working on this.

Strakata, 2020-09-17:
This would be a major improvement!
Unfortunately I don't have the knowledge to help, apart from testing Beta releases with an OCPP capable charger.

chuck-h, 2021-04-24:
For a home automation interface that needs only visibility into SteVe status (e.g. has charging session completed?) consider executing read-only access to SteVe's SQL database. No modification to SteVe code is required.
To support external control of charging operations (e.g. dynamic current limit) a limited API may be sufficient #552

juherr, 2026-02-02:
Home Assistant mainly focuses on metrics and actions, and it is capable of hosting containers on HAOS.

For this scope, the current Powerfill / SteVe API should be sufficient. An optional SteVe container for Home Assistant could also be provided.

For more advanced scenarios, such as energy-usage automation or dynamically updating charging profiles (related to #92), a more event-driven mechanism like webhooks or MQTT will be required.

That said, webhooks from Powerfill to Home Assistant are not a strict requirement, as many Home Assistant instances are not accessible from outside the local network.

For reference, a SteVe container for Home Assistant already exists:
https://github.com/erik73/addon-steve

In that context, SteVe could also be considered a viable alternative to:
https://github.com/lbbrhzn/ocpp

I hope this helps clarify the landscape.