steve-community/steve #118: ocpp-jaxb-0.0.1 missing

war-bln, 2019-03-12:
Hi there,

I've tried to package / install steve (3.2.0 and master) and had an issue due unavailability of "ocpp-jaxb-0.0.1". 
It is not possible to download pom.xml and jar. Could you please provide the missing dependency through your public nexus?

Thanks in regard,
war

PS: see maven output below:

[INFO] --------------------------------[ jar ]---------------------------------
[INFO] Downloading from nexus-public: http://(...)/nexus/content/groups/public/de/rwth/idsg/ocpp-jaxb/0.0.1/ocpp-jaxb-0.0.1.pom
[WARNING] The POM for de.rwth.idsg:ocpp-jaxb:jar:0.0.1 is missing, no dependency information available
[INFO] Downloading from nexus-public: http://(...)/nexus/content/groups/public/de/rwth/idsg/ocpp-jaxb/0.0.1/ocpp-jaxb-0.0.1.jar

goekay, 2019-03-12:
i just cleaned my maven cache and tried to build with maven again. worked without any problems.

adi64, 2019-03-12:
Maybe you need to build https://github.com/RWTH-i5-IDSG/ocpp-jaxb locally?

goekay, 2019-03-12:
> Maybe you need to build https://github.com/RWTH-i5-IDSG/ocpp-jaxb locally?

my last commit has some major changes and i bumped the version, which i don't use in steve yet. if you want to do it, you should checkout the second last commit and run `mvn install` on it.

goekay, 2019-03-12:
@war-bln did you modify maven settings or pom.xml? steve's pom.xml defines two repos: `central` and `TreibhausNexus`. `TreibhausNexus` is ours with the `ocpp-jaxb` package. your maven output only references `nexus-public`, which is neither `central` nor `TreibhausNexus`.

war-bln, 2019-03-12:
I've found the solution: had to download the additional sources from https://github.com/RWTH-i5-IDSG/ocpp-jaxb

closed 2019-03-12 by war-bln