steve-community/steve #119: not able to connect steve websocket endpoint through remote machine

javatoast, 2019-03-14:
Hello,

After running steve on local machine, i can send the bootstrap message, connect, hearbeat from local machine. Even i am able to send the heartbeat from java client on local box. I mean everything works perfect when server and client machine are same.

Problem is when i try to send heartbeat from other machine using java client or html client with server IP address it fails on connection. Any pointers.

goekay, 2019-03-14:
i don't know how to help you, since you did not give any details about your setup and network configuration. can you at least provide a stack trace of excepion (if available) of connections failure?

on the other hand, this seems like a problem about your environment and has little to do with steve itself.

goekay, 2019-04-08:
closing due to inactivity.

closed 2019-04-08 by goekay