steve-community/steve #120: Error trying to start application.

romfrolov, 2019-03-16:
I have successfully built application with Maven, but when I try to launch application I get following error log:

```
[INFO ] 2019-03-16 19:44:22,445 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2019-03-16 19:44:22,550 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2019-03-16 19:44:22,570 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2019-03-16T16:44:22.556Z
[INFO ] 2019-03-16 19:44:22,596 org.eclipse.jetty.util.log - Logging initialized @926ms to org.eclipse.jetty.util.log.Slf4jLog
[INFO ] 2019-03-16 19:44:22,755 org.eclipse.jetty.server.Server - jetty-9.4.z-SNAPSHOT; built: 2018-11-14T21:20:31.478Z; git: c4550056e785fb5665914545889f21dc136ad9e6; jvm 11.0.2+9
[WARN ] 2019-03-16 19:44:22,795 org.eclipse.jetty.xml.XmlParser - FATAL@null line:44 col:26 : org.xml.sax.SAXParseException; lineNumber: 44; columnNumber: 26; The element type "servlet-class" must be terminated by the matching end-tag "</servlet-class>".
[WARN ] 2019-03-16 19:44:22,795 org.eclipse.jetty.webapp.WebAppContext - Failed startup of context o.e.j.w.WebAppContext@117e0fe5{/steve,jar:file:/Users/romanfrolov/Desktop/steve/target/steve.jar!/webapp,UNAVAILABLE}
org.xml.sax.SAXParseException: The element type "servlet-class" must be terminated by the matching end-tag "</servlet-class>".
	at com.sun.org.apache.xerces.internal.util.ErrorHandlerWrapper.createSAXParseException(ErrorHandlerWrapper.java:204) ~[?:?]
	at com.sun.org.apache.xerces.internal.util.ErrorHandlerWrapper.fatalError(ErrorHandlerWrapper.java:178) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:400) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:327) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLScanner.reportFatalError(XMLScanner.java:1471) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanEndElement(XMLDocumentFragmentScannerImpl.java:1683) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(XMLDocumentFragmentScannerImpl.java:2881) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.next(XMLDocumentScannerImpl.java:605) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLNSDocumentScannerImpl.next(XMLNSDocumentScannerImpl.java:112) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanDocument(XMLDocumentFragmentScannerImpl.java:534) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:888) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:824) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.parse(AbstractSAXParser.java:1216) ~[?:?]
	at com.sun.org.apache.xerces.internal.jaxp.SAXParserImpl$JAXPSAXParser.parse(SAXParserImpl.java:635) ~[?:?]
	at com.sun.org.apache.xerces.internal.jaxp.SAXParserImpl.parse(SAXParserImpl.java:324) ~[?:?]
	at org.eclipse.jetty.xml.XmlParser.parse(XmlParser.java:255) ~[steve.jar:?]
	at org.eclipse.jetty.webapp.Descriptor.parse(Descriptor.java:55) ~[steve.jar:?]
	at org.eclipse.jetty.webapp.WebDescriptor.parse(WebDescriptor.java:212) ~[steve.jar:?]
	at org.eclipse.jetty.webapp.MetaData.setWebXml(MetaData.java:194) ~[steve.jar:?]
	at org.eclipse.jetty.webapp.WebXmlConfiguration.preConfigure(WebXmlConfiguration.java:60) ~[steve.jar:?]
	at org.eclipse.jetty.webapp.WebAppContext.preConfigure(WebAppContext.java:506) ~[steve.jar:?]
	at org.eclipse.jetty.webapp.WebAppContext.doStart(WebAppContext.java:544) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:108) ~[steve.jar:?]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[steve.jar:?]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.doStart(GzipHandler.java:406) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:117) ~[steve.jar:?]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[steve.jar:?]
	at org.eclipse.jetty.server.Server.start(Server.java:415) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:108) ~[steve.jar:?]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[steve.jar:?]
	at org.eclipse.jetty.server.Server.doStart(Server.java:382) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:144) ~[steve.jar:?]
	at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:37) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.start(Application.java:57) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.main(Application.java:46) ~[steve.jar:?]
[INFO ] 2019-03-16 19:44:22,825 org.eclipse.jetty.server.AbstractConnector - Started ServerConnector@40dd3977{HTTP/1.1,[http/1.1]}{127.0.0.1:8080}
[ERROR] 2019-03-16 19:44:22,825 de.rwth.idsg.steve.SteveProdStarter - Exception happened
org.xml.sax.SAXParseException: The element type "servlet-class" must be terminated by the matching end-tag "</servlet-class>".
	at com.sun.org.apache.xerces.internal.util.ErrorHandlerWrapper.createSAXParseException(ErrorHandlerWrapper.java:204) ~[?:?]
	at com.sun.org.apache.xerces.internal.util.ErrorHandlerWrapper.fatalError(ErrorHandlerWrapper.java:178) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:400) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:327) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLScanner.reportFatalError(XMLScanner.java:1471) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanEndElement(XMLDocumentFragmentScannerImpl.java:1683) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(XMLDocumentFragmentScannerImpl.java:2881) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.next(XMLDocumentScannerImpl.java:605) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLNSDocumentScannerImpl.next(XMLNSDocumentScannerImpl.java:112) ~[?:?]
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanDocument(XMLDocumentFragmentScannerImpl.java:534) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:888) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:824) ~[?:?]
	at com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141) ~[?:?]
[... 13 lines cut ...]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[steve.jar:?]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.doStart(GzipHandler.java:406) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:117) ~[steve.jar:?]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[steve.jar:?]
[... cut ...]
	at de.rwth.idsg.steve.Application.main(Application.java:46) ~[steve.jar:?]
[INFO ] 2019-03-16 19:44:22,833 org.eclipse.jetty.server.AbstractConnector - Stopped ServerConnector@40dd3977{HTTP/1.1,[http/1.1]}{127.0.0.1:8080}
[INFO ] 2019-03-16 19:44:22,834 org.eclipse.jetty.server.handler.ContextHandler - Stopped o.e.j.w.WebAppContext@117e0fe5{/steve,jar:file:/Users/romanfrolov/Desktop/steve/target/steve.jar!/webapp,UNAVAILABLE}
```
---

MacOS Mojave 10.14
openjdk 11.0.2 2019-01-15
mysql  Ver 14.14 Distrib 5.7.25, for osx10.14 (x86_64) using  EditLine wrapper

goekay, 2019-03-16:
did you modify something? or did you build the project as is in git branch?

romfrolov, 2019-03-16:
@goekay I modified this line: `hc.addDataSourceProperty(PropertyDefinitions.PNAME_useSSL, false);` in **BeanConfiguration.java** and `<jdbcUrl>jdbc:mysql://${db.ip}:${db.port}/${db.schema}?allowPublicKeyRetrieval=true&amp;useSSL=false&amp;serverTimezone=UTC</jdbcUrl>` in **pom.xml**.

commit dd407b7d (2019-03-17): update jetty version (Closes #120)
closed 2019-03-17 by goekay