steve-community/steve #124: Build of 3.2.0 fails on fresh Raspberry Pi installation

V2G-UK, 2019-03-26:
This is undoubtedly a duplicate of #101. For chapter and verse please see:

http://www.V2G-EVSE.com/2018/04/16/ev-charging-station-administration-using-ocpp-1-6-with-steve/#comment-1214

However it does raise the question of why you have chosen to force SteVe users to install more recent versions of a number of prerequisites than are readily available in the  "latest stable" version of their chosen OS distribution?

csamsel, 2019-03-26:
It's impossible to cover all distributions in terms of MySQL and Java versions. The release date between the MySQL version that comes with "latest" Debian stable and with the latest Fedora versions is 8 years apart (Debian 5.5: 2010, Fedora 5.8: 2018)

Development takes place for current software versions, not outdated ones which arent supported upstream.

Feel free to use for example Ubuntu 18.04 LTS instead.

goekay, 2019-03-26:
i really make an effort to keep all the libraries up-to-date without breaking things for users. the only exceptions are unfortunately jdk and mysql, which are external dependencies, and i try to be mindful in these cases.

mysql version bump was out of necessity. i wanted to use a feature in a database view, which was not supported by the previously used version. so we had to switch mysql 5.7.7, which was the first version that supported this feature and was released at [2015-04-08](https://dev.mysql.com/doc/relnotes/mysql/5.7/en/news-5-7-7.html). _4 years ago!_ so... not really bleeding edge, is it? if your chosen distro is so outdated that it uses even an earlier mysql version, i would suggest to use something else, because who knows what other critical outdated libraries it uses with potential security issues.

V2G-UK, 2019-03-26:
@csamsel - We have Ubuntu 18.04 in house, amongst other things. I was merely trying to solve a problem reported by a reader of one of my blogs.

@goekay - It's not my "chosen distro", see above. Whilst a Raspberry Pi running Raspbian Stretch certainly wouldn't be own my first choice for a SteVe platform, there are an awful lot of them out there in the wild!

@Both - Thanks for your swift responses.

goekay, 2019-03-27:
>  there are an awful lot of them out there in the wild!

i believe that, Jim. but on the other extreme of the spectrum, we had issues with people reporting that steve is not compatible with mysql 8.0, since it made some major breaking changes. so there are people living on the edge, too. and here i was thinking to migrate to mysql 8.0 because it is a huge step forward for mysql (but i probably will postpone it after your feedback).

regarding java: jdk 8 is officially in maintenance mode. oracle dropped supporting it. openjdk (with the help of redhat) promises to provide critical bug and security fixes until 2023. but they do not promise back-porting important new features.

i am sorry that steve does not work out-of-the-box for some distros. it's hard to find a happy medium ground to cover the use cases of all users.

V2G-UK, 2019-03-27:
I'm beginning to wonder why I wandered down this particular rabbit hole!

Now attempting to build 3.2.0 on my HP "CORE i7 8th gen" powered laptop running native Ubuntu 18.04 I am told:

[image]

http://www.V2G-EVSE.com/2018/04/16/ev-charging-station-administration-using-ocpp-1-6-with-steve/#comment-1219

goekay, 2019-03-27:
this is a problem with ubuntu, not steve. as you can see from your console log, even though you want to install openjdk-11-jdk, the repos provide you jdk10. the maintainers named the package that way, because they wanted to switch to jdk11 as soon as it was released. but it did not happen. and it is very confusing, i agree!

further info and help: https://www.linuxuprising.com/2019/01/how-to-install-openjdk-11-in-ubuntu.html

V2G-UK, 2019-03-27:
Since it's what we use on the "embedded" end of things I actually used Azul Zulu 11.0.2:

https://www.azul.com/downloads/zulu/zulu-linux/

That was sufficient to get SteVe 3.2.0 up and running for me.

closed 2019-04-08 by goekay