steve-community/steve #125: Remote Start Transaction Energy limit

daniel8702, 2019-04-05:
Hello, 
Is it possible to send to the OCCP client using the RemoteStartTransaction the energy value, after which the charger will automatically stop charging?

goekay, 2019-04-05:
ocpp spec defines it, so... yes. but it is not implemented in steve yet. it is a little bit more involved than just sending an energy value, though. ocpp spec uses the term _charging profile_, which fully describes the time periods, charing rates and other details of charging sessions.

daniel8702, 2019-04-08:
do you plan to extend the system with such functionality? I think that this is a very interesting possibility to send a charging profile to the charger, on the basis of which the charger would automatically finish loading.

goekay, 2019-04-08:
> do you plan to extend the system with such functionality? 

i remember this operation being cumbersome to implement, which is why i left it out initially. i can look into it when i have time.

> ... possibility to send a charging profile to the charger

this is already possible, but in a slightly verbose way, i guess. after remote start tansaction, the station should send a Start Transaction message. during a transaction you can use the operation Set Charging Profile to control the charging behaviour.

billywatsy, 2022-08-11:
@daniel8702 @goekay , 

l want transaction id xxxxx to stop after charging xxxx KW or a hack around it.

{
	"connectorId": 0,
	"idTag": "xxxxxxxxxxxx",
	"chargingProfile": {
		"chargingProfileId": 10,
		"transactionId": null,
		"stackLevel": 0,
		"chargingProfilePurpose": "TxProfile",
		"chargingProfileKind": "Relative",
		"recurrencyKind": "Daily",
		"validFrom": "2022-02-18T10:18:46.000Z",
		"validTo": "2022-08-18T10:18:46.000Z",
		"chargingSchedule": {
			"duration": 4200,
			"startSchedule": "2022-08-18T10:18:46.000Z",
			"chargingRateUnit": "W",
			"chargingSchedulePeriod": [
				{
					"startPeriod": 0,
					"limit": 50
				},
				{
					"startPeriod": 120,
					"limit": 0
				}
			] 
		}
	}
}

will setting the last value to zero work

[image]

artemboyko43, 2025-01-16:
Any news here? At least clarification will be good here how to stop charging by **charging profile**

goekay, 2025-09-14:
sending a charging profile within a RemoteStartTransaction was implemented in https://github.com/steve-community/steve/pull/1809 recently.

closed 2025-09-14 by goekay