steve-community/steve #126: Scheduled auto reboot

goRaspy, 2019-04-05:
Related to my experience it is very useful to regularly reboot stations.

I think it should be a very nice to have a feature for Steve to implement an "auto reboot" function.

In my opinion, Steve may offer a GUI to let user plan a global reboot (hard reset) of connected stations. Ideally we may be able to choose a period (every hour / day / week) and time. Obviously Steve may only reboot stations with a connector status "available" on each connector.( for some cases, reboot can fix problem related to transaction "never close" so Steve may not look at open transactions to choose concerned stations).

I hope my English is understandable ^^

Gior80, 2019-04-05:
Why don’t do that firmware side ?

goRaspy, 2019-04-05:
I do not understand. Steve is a backend compatible with a large amount of
charging stations. Not a charger side solution.

giorgio80, 2019-04-05:
let me better explain. Reset is already a part of ocpp protocol and already implemented in task you can run over station(as you say). what I'm saying is that if a firmware has a good ocpp implementation it should not hangs or have a good workaround routines.  This is why I don't see a global reset as a way to patch bad firmwares

goRaspy, 2019-04-05:
Sorry it is more clear for me now.
I completly agree with you but it is a very theorical point of view.
In real life with serious experience with différent manufacturers. I can
confim the real utility of this feature.

giorgio80, 2019-04-05:
so choose right station  then :)

goekay, 2019-04-05:
> Reset is already a part of ocpp protocol ...

it is indeed implemented and you can trigger it for multiple stations _manually_. as i understand, @goRaspy wants an automated process that triggers this operation at scheduled times without human intervention.

giorgio80, 2019-04-05:
@goRaspy , leaving out a little bit from the topic,  but could u share good and bad hanging experience with stations ? I mean more stable and more unstable

goekay, 2019-04-05:
@giorgio80 we had multiple github issues with reports due to misbehaving stations. please take a look at them. for example, https://github.com/RWTH-i5-IDSG/steve/issues/116, https://github.com/RWTH-i5-IDSG/steve/issues/94 and later comments in https://github.com/RWTH-i5-IDSG/steve/issues/97

goRaspy, 2019-04-05:
Yes @goekay this is that ^^

@giorgio80 , honnestly I think most of "big" manufacturers have now a mature OCCP implementation for "classic" commands. Most recent commands like charging profile still have ambiguous and exotics implementations.
Only big point, it should be forbidden to sell soap charging stations by 2019.

But there is always something "bad" who can happen with an IOT online 24/7.

Gior80, 2019-04-05:
Ou was not knowing about this regulamentation of soap station from 2019 , it is law ? @goakay we’re making bar chats . Need to continue private ?

goRaspy, 2019-04-05:
No it is a joke :)
But there is a huge difference between SOAP and JSON.

Gior80, 2019-04-05:
Soap sucks 😀 I do js firmware whole life

RedPeace, 2021-11-19:
Hi,

unfortunately we have also the problem with "ghost transactions". Some stations don't close her transactions. And after some time charging isn't possible because of too many "open" transactions. After i kill the transactions manually via web gui charging is possible again. An automatic closing feature would be great!

Question: Would a reboot of the steve server close this ghost transactions?

Regards
Gero

goekay, 2025-12-17:
> unfortunately we have also the problem with "ghost transactions". Some stations don't close her transactions. And after some time charging isn't possible because of too many "open" transactions. [...]

you are trying to fix the symptom (or rather: improve the workaround we provide) instead of fixing the root cause ;). there should not be too many open transactions in a healthy station-backend relationship. many things changed and improved over the years on station side. i would like to imagine that current hardware and software/firmware are more stable and do not exhibit the issues of the earlier era.

> Question: Would a reboot of the steve server close this ghost transactions?

no, not really, because unrelated.

juherr, 2025-12-18:
@goekay Even though hardware and firmware quality have significantly improved over time, legacy devices are still part of real-world deployments, and missing StopTransaction messages can still occur for various reasons.

You’re absolutely right that the root cause does not lie in the CSMS itself. That said, a “CSMS-friendly” approach can help operators recover from these situations once they happen in production.

For instance, the Steve UI could offer a manual “Stop transaction” action, opening a small form to capture the stop timestamp and meter value.

Another possible improvement could be to automatically close an ongoing transaction when a new StartTransaction is received on the same connector, using sensible defaults or operator-defined rules.

These kinds of mechanisms would not mask device issues, but they could significantly improve day-to-day operability when dealing with legacy or unstable hardware.

goekay, 2025-12-18:
> I'm stuck with an old version without that feature.

this feature has been in steve for a _very long_ time. you must be using a really old version. if i may ask: why?