steve-community/steve #127: Illegal reflective access by org.springframework.cglib.core.ReflectUtils

TechOverflow, 2019-04-08:
Would anyone happen to know how to solve these errors? I would guess they are due to incorrect dependencies.

`WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.`

```
Starting...WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.springframework.cglib.core.ReflectUtils$1 (file:/opt/steve/target/steve.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of org.springframework.cglib.core.ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

goekay, 2019-04-08:
the warnings happen because the approach how to use reflection changed after jdk 9 and it takes a while for library/framework developers to migrate to the new approach. i am currently on HEAD of the main git branch and i see these warnings only because of jooq but not spring. spring version update with commit fa2bfa76 must have fixed it. jooq will fix this with version 3.12.

TechOverflow, 2019-04-08:
Thanks. Yeah, I went through the version numbers in my pom and had an old one.
Now I get the jooq warning, probably the one you mention

`WARNING: Illegal reflective access by org.jooq.tools.reflect.Reflect`

But as long as this is a known "issue", all is well. It doesn't have any effect on functionality, as far as I can tell.

closed 2019-04-08 by goekay