steve-community/steve #130: Compilation error with END header not found

CristobalBL, 2019-04-16:
Hi, 

I cannot build steve (release 3.2.0 and clone from master). 

I try with maven 3.3.9 and 3.5.2. 

When i run mvn package: 

- For all dependences from tribhaus.informatik/nexus/repository ... :
`[WARNING] Could not validate integrity of download from https://treibhaus.informatik.rwth-aachen.de/nexus/repository/maven-public/de/rwth/idsg/ocpp-jaxb/0.0.1/ocpp-jaxb-0.0.1.pom: Checksum validation failed, expected <!DOCTYPE but is 86be636d7b854886c4abc44181a078a067632cd6
[WARNING] Checksum validation failed, expected <!DOCTYPE but is 86be636d7b854886c4abc44181a078a067632cd6 for https://treibhaus.informatik.rwth-aachen.de/nexus/repository/maven-public/de/rwth/idsg/ocpp-jaxb/0.0.1/ocpp-jaxb-0.0.1.pom
 ...
Downloaded: https://treibhaus.informatik.rwth-aachen.de/nexus/repository/maven-public/de/rwth/idsg/ocpp-jaxb/0.0.1/ocpp-jaxb-0.0.1.pom (28 KB at 1.6 KB/sec)
[WARNING] The POM for de.rwth.idsg:ocpp-jaxb:jar:0.0.1 is invalid, transitive dependencies (if any) will not be available, enable debug logging for more details`

- First error message:
`[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] error reading /home/wallbox/.m2/repository/de/rwth/idsg/ocpp-jaxb/0.0.1/ocpp-jaxb-0.0.1.jar; zip END header not found ...`

goekay, 2019-04-16:
we are aware of that. sorry for the problem. unfortunately, our nexus repository is down and we are working on it.

goekay, 2019-04-16:
the problem has been fixed. our nexus repo is up and running.

closed 2019-04-16 by goekay