steve-community/steve #131: Compile issue: .FileNotFoundException

shengxiang200788, 2019-04-25:
java -version:
java version "12.0.1" 2019-04-16
Java(TM) SE Runtime Environment (build 12.0.1+12)
Java HotSpot(TM) 64-Bit Server VM (build 12.0.1+12, mixed mode, sharing)

 mvn -version:
Maven home: E:\software\apache-maven-3.5.3
Java version: 12.0.1, vendor: Oracle Corporation
Java home: E:\software\jdk-12.0.1
Default locale: zh_CN, platform encoding: GBK
OS name: "windows 10", version: "10.0", arch: "amd64", family: "windows"

Got the followings:
[DEBUG] Generated E:\workshop\java\steve-steve-3.2.0\target\classes\org\apache\jsp\WEB_002dINF\views\_00_002dcontext_jsp.java total=355 generate=56 validate=296
[DEBUG] Compiled E:\workshop\java\steve-steve-3.2.0\target\classes\org\apache\jsp\WEB_002dINF\views\_00_002dcontext_jsp.java 737ms
[DEBUG] Generating web mapping for file \WEB-INF\views\00-context.jsp using compilation context org.apache.jasper.JspCompilationContext@450d10cb
[DEBUG] Processing file: \WEB-INF\views\00-cp-multiple.jsp
[DEBUG] /WEB-INF/views/00-cp-multiple.jsp is out dated, compiling...
[DEBUG] Generated E:\workshop\java\steve-steve-3.2.0\target\classes\org\apache\jsp\WEB_002dINF\views\_00_002dcp_002dmultiple_jsp.java total=143 generate=48 validate=94
[WARNING] Compilation error
java.io.FileNotFoundException: E:\workshop\java\steve-steve-3.2.0\target\classes\org\apache\jsp\WEB_002dINF\views\_00_002dcp_002dmultiple_jsp.class

goekay, 2019-04-25:
> java version "12.0.1" 2019-04-16

i suspect that jdk 12 is the problem. please try with jdk 11.

goekay, 2019-05-07:
> i suspect that jdk 12 is the problem. please try with jdk 11.

i successfully built with jdk 12 and therefore cannot reproduce the problem. since the issuer did not provide any follow-up messages and details, i am closing this issue due to inactivity.

closed 2019-05-07 by goekay