steve-community/steve #132: Firmware update uses local time

lategoodbye, 2019-04-25:
Hi,
i'm using Steve 3.2.0. At URL /steve/manager/operations/v1.6/UpdateFirmware i need to provide a retrieve date/time for the actual firmware update. The form uses a DateTimePicker which is very helpful and provides the value in localtime. Unfortunately the actual UpdateFirmware.Req uses the localtime and wrongly declare it as Zulu time:

Localtime: 2019-04-25 9:14 CEST

[INFO ] 2019-04-25 07:22:46,325 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=tarragon_leipzig, sessionId=2fb8eecf] Sending: [2,"9f5088dc-e063-4133-a7da-f437fd0b4a52","UpdateFirmware",{"retrieveDate":"2019-04-25T09:14:00.000Z","location":"http://download.i2se.com/ApplicationNotes/an4_rev5.pdf"}]

As a result i'm not able to specify a retrieve datetime sooner than two hours.

Personally i see two solutions.

Option 1) the retrieve date form / dateTimePicker allows a value in Zulu time

Option 2) Steve should properly convert the localtime browser time into Zulu time

goekay, 2019-04-25:
> The form uses a DateTimePicker which is very helpful and provides the value in localtime. Unfortunately the actual UpdateFirmware.Req uses the localtime and wrongly declare it as Zulu time

currently, steve understands only one time zone, which is by default UTC (or as you call it zulu time). the dates and times presented in browser and entered in the forms are all in this time zone. there is already one feature request #96 to improve this situation. i yet have to find the time to work on it. when done, the solution will correspond to your option 2.

> As a result i'm not able to specify a retrieve datetime sooner than two hours.

unfortunately, the cause of the problem is the range restriction in DateTimePicker. since the backend validates the input values anyways, i will disable the restriction in DateTimePicker, such that the user will be able to select any date and time in browser.

closed 2019-04-25 by goekay

rdc-Green, 2020-11-16:
> Hi,
> i'm using Steve 3.2.0. At URL /steve/manager/operations/v1.6/UpdateFirmware i need to provide a retrieve date/time for the actual firmware update. The form uses a DateTimePicker which is very [...]

@lategoodbye , can you tell me which FTP server you used to send firmware to charge point? I'm trying to send firmware update using Filezilla with Ubuntu on Azure VM. It connects on Filezilla and FTPtest.net, but still doesn't download the firmware. Thanks

commit 125f9b8b (2019-04-25): disable datetime picker range restriction

this closes #132