steve-community/steve #134: No answer starttransaction.conf

goRaspy, 2019-05-07:
In theory, it is impossible to receive a "StartTransaction" including and invalid idtag. In reality that can happen in different cases.

In this situation Steve never answer the StartTransaction.conf and the CP stay waiting (and stuck) for an answer.
[image]

I am pretty sure that is the explanation of "hikari pool" errors, related by some Steve users (including me).

By my opinion, it should be important to catch this exception and answer "something" to the charging station. I think a good solution may be to store the transaction in database with a "default" IDTAG (like XXXXXX). This should be interesting to "unfreeze" the station by answering a transactionId without have to manage the list of "unknown" Idtag. 

For me the protocol is not clear on this point. But it should be appropriate to answer an "invalid status" to the Idtag status in StartTrasaction.conf

Exemple :

CP to CS :
StartTransaction : 
"StartTransaction",{"connectorId":1,"idTag":"XAXAXA","timestamp":"2019-05-07T05:52:05Z","meterStart":0}]

Where XAXAXA is Invalid

CS to CP
{"transactionId":12345,"idTagInfo":{"status":"Invalid"}]

goekay, 2019-05-07:
> In theory, it is impossible to receive a "StartTransaction" including and invalid idtag. 

this is not true. after people pointed this case out, i implemented a logic that inserts invalid tags and transaction some time ago, such that no data is lost. you can see it here in the flow of operations: https://github.com/RWTH-i5-IDSG/steve/blob/b42860dfd3a0189b08bc311c104843cc1056712b/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java#L164-L214

therefore, what you wish should happen is actually the implemented logic. therefore, i cannot fully understand your point. can you be more clear? and post for example the logs and exceptions?

goRaspy, 2019-05-07:
I am really sorry, I am on a build older than on 14 Dec 2018
Will try to deploy the latest version and make a test.

Many thanks for your precise and fast answer (as usual).

closed 2019-05-07 by goekay

goekay, 2019-05-07:
this specific feature was added with commit 3be0002acef7452ea0023532fdd9192ae9fcb1b3 after the report in issue #13