steve-community/steve #135: No protocol (OCPP version)

solodoit, 2019-05-07:
I add a chargeBox, but there no protocol 
[image]
[image]

goekay, 2019-05-07:
the ocpp websocket spec requires the following in section "3.1.2. OCPP version" of the document "OCPP-J 1.6 Specification": 

> The exact OCPP version MUST be specified in the Sec-Websocket-Protocol field.

since your charging station does not specify one of the allowed values, steve does not accept the websocket connection attempt.

closed 2019-05-24 by goekay

rathourcode, 2023-04-19:
> Sec-Websocket-Protocol field.

hai goekay 

can you please tell how to specify ocppp version so that my websocket coonection works.