steve-community/steve #136: Timezone issue

TechOverflow, 2019-05-10:
I am not sure if this is a SteVe issue or rather the charger's. According to the log, it seems to me that the charger got the current time, but did not change it fast enough.
The charger was connected to a different endpoint before this.

```
Connection is established

Received: [2, "135561825", "BootNotification",...

The chargebox 'TEST1' is registered and its boot acknowledged.

Sending: [3,"135561825",{"status":"Accepted","currentTime":"2019-05-10T14:05:43.546Z","interval":300}]

Received: [2, "156167709", "StatusNotification", {"connectorId": 1, "status": "Available", "errorCode": "NoError", "info": "Status Update", "timestamp": "2019-05-10T16:05:33Z", "vendorId": "Chago"}]

Sending: [3,"156167709",{}]

Received: [2, "254045070", "StatusNotification", {"connectorId": 0, "status": "Available", "errorCode": "NoError", "info": "Status Update", "timestamp": "2019-05-10T16:05:33Z", "vendorId": "Chago"}]

Sending: [3,"254045070",{}]

Received: [2, "218177795", "StatusNotification", {"connectorId": 0, "status": "Available", "errorCode": "NoError", "info": "Status Update", "timestamp": "2019-05-10T14:05:43Z", "vendorId": "Chago"}]

Sending: [3,"218177795",{}]
```

As you can see, the reply to the BootNotification sets the current UTC time, but the charger sent two StatusNotifications with the wrong time (con 1, con 0). 

Is it worth doing anything about this or is this minor enough to ignore?
The log times are UTC+2, shouldn't the currentTime be +2 as well?

goekay, 2019-05-10:
> Is it worth doing anything about this or is this minor enough to ignore? The problem is that there is now a record showing con 0, 1 are available at 16:05 so any any connector status update will be moved back 2 hours or until UTC time is 16:05+.

it is not minor, this is indeed problematic, i agree with your assumption that the station receives the current time but probably does not react fast enough. i would not add a workaround in steve though, since this seems to be a fault of the station. the station software should be fixed by the manufacturer.

moreover, i cannot imagine that the internal clock of the station can be so much off (if it is connected to internet and ntp is running) and can produce a timestamp like "2019-05-10T16:05:33Z" (i.e. in utc timezone but... the clock is 2 hours ahead?). this might be an indicator of something bigger like the station not understanding the concept of timezones but putting a "Z" at the end anyways. therefore, maybe the internal clock was in local timezone at first (hence the 16:05:33). after receiving the timestamp from steve, it overriden the time and thinks that the time is 14:05:43 now.

>The log times are UTC+2, shouldn't the currentTime be +2 as well?

unfortunately, i could not manage the logging framework to respect the timezone that we set in [SteveConfiguration](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java#L28).

TechOverflow, 2019-05-10:
Interesting. I'll have a look at station settings etc.

> i cannot imagine that the internal clock of the station can be so much off

I think this is due to the station being connected to a different server before I changed it. The charger seems to have saved the time the last server gave it.

In regards to the timezones, and the reason why I was looking at them in my testing - ID tags have expiry date/time. This is a good feature, but something that the OCPP standard seemingly overlooked is a way to deal with expiry during a transaction. The StartTransaction reply includes the expiry date/time, but this seems to only update local cache (on station side). There is no function that would block the transaction while it is happening.

This is problematic because, in theory, a transaction started 10.5.2019 at 15:00 for an ID tag that has expiry 10.5.2019 15:30, the user is able to charge as long as they don't end the transaction (even many hours or days).

Would it be worth adding a feature that would force end transactions upon expiry date/time? Since I don't see OCPP implementing this any time soon.

goekay, 2019-05-10:
> This is problematic because, in theory, a transaction started 10.5.2019 at 15:00 for an ID tag that has expiry 10.5.2019 15:30, the user is able to charge as long as they don't end the transaction (even many hours or days).

it looks like that enforcing such a restriction is the responsibility of the station, if the ocpp spec mentions it. i don't see the necessity of an additional function/operation.

> Would it be worth adding a feature that would force end transactions upon expiry date/time?

no. this would be stepping outside the spec, if it is not clearly mentioned in the spec.

TechOverflow, 2019-05-10:
I have updated the firmware and it seems that the date/time problem is gone even when I intentionally gave the charger a different time by connecting to a different server, then back.

```
Received: [2, "245734615", "BootNotification",...

The chargebox 'TEST01' is registered and its boot acknowledged.

Sending: [3,"245734615",{"status":"Accepted","currentTime":"2019-05-10T19:14:43.424Z","interval":300}]

Received: [2, "184758027", "StatusNotification", {"connectorId": 0, "status": "Available", "errorCode": "NoError", "info": "Status Update", "timestamp": "2019-05-10T19:14:43Z", "vendorId": "Chago"}]

Sending: [3,"184758027",{}]

Received: [2, "221866290", "StatusNotification", {"connectorId": 1, "status": "Charging", "errorCode": "NoError", "info": "Status Update -Type2-", "timestamp": "2019-05-10T19:14:43Z", "vendorId": "Chago"}]

Sending: [3,"221866290",{}]

Received: [2, "190186238", "StatusNotification", {"connectorId": 2, "status": "Unavailable", "errorCode": "NoError", "info": "Status Update", "timestamp": "2019-05-10T19:14:43Z", "vendorId": "Chago"}]

Sending: [3,"190186238",{}]
```

closed 2019-05-10 by TechOverflow