steve-community/steve #138: Error on database migration

victormunoz, 2019-05-22:
This is the error I get when running mvn install in a fresh installation:

```
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:5.2.4:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: 
[ERROR] Migration V0_8_5__update.sql failed
[ERROR] -----------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1215
[ERROR] Message    : Cannot add foreign key constraint
[ERROR] Location   : C:\Users\Victor\Desktop\svn\Ocpp\Emser\src\main\resources\db\migration\V0_8_5__update.sql (C:\Users\Victor\Desktop\svn\Ocpp\Emser\src\main\resources\db\migration\V0_8_5__update.sql)
[ERROR] Line       : 29
[ERROR] Statement  : ALTER TABLE `charge_box`
[ERROR] ADD address_pk INT DEFAULT NULL,
[ERROR] ADD CONSTRAINT FK_charge_box_address_apk
[ERROR] FOREIGN KEY (address_pk) REFERENCES address (address_pk) ON DELETE SET NULL ON UPDATE NO ACTION
```

The database is mysql 5.7.14

goekay, 2019-05-22:
i just tested building and starting steve with a fresh mysql 5.7.14 installation on windows 7. it worked. i cannot reproduce the problem.

goekay, 2019-06-10:
i am closing this issue due to inactivity.

closed 2019-06-10 by goekay