steve-community/steve #139: MySQL and Java are not using the same time zone

victormunoz, 2019-05-22:
I am getting this error at execution time.

```
Starting.....WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.jooq.tools.reflect.Reflect (file:/C:/Users/Victor/Desktop/svn/Ocpp/Emser/target/libs/jooq-3.11.9.jar) to constructor java.lang.invoke.MethodHandles$Lookup(java.lang.Class)
WARNING: Please consider reporting this to the maintainers of org.jooq.tools.reflect.Reflect
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
........ FAILED!
```

The log says this (three times):

```
[ERROR] 2019-05-22 18:33:01,037 de.rwth.idsg.steve.SteveProdStarter - Exception happened
de.rwth.idsg.steve.SteveException: MySQL and Java are not using the same time zone. Java offset in seconds (0) != MySQL offset in seconds (7200)
	at de.rwth.idsg.steve.utils.DateTimeUtils.checkJavaAndMySQLOffsets(DateTimeUtils.java:89) ~[steve.jar:?]
	at de.rwth.idsg.steve.config.BeanConfiguration.afterStart(BeanConfiguration.java:150) ~[steve.jar:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.doInvoke(ApplicationListenerMethodAdapter.java:259) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.processEvent(ApplicationListenerMethodAdapter.java:179) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.onApplicationEvent(ApplicationListenerMethodAdapter.java:142) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.doInvokeListener(SimpleApplicationEventMulticaster.java:172) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.invokeListener(SimpleApplicationEventMulticaster.java:165) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.multicastEvent(SimpleApplicationEventMulticaster.java:139) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:398) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:355) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.finishRefresh(AbstractApplicationContext.java:882) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:549) ~[spring-context-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.web.context.ContextLoader.configureAndRefreshWebApplicationContext(ContextLoader.java:400) ~[spring-web-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.web.context.ContextLoader.initWebApplicationContext(ContextLoader.java:291) ~[spring-web-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.springframework.web.context.ContextLoaderListener.contextInitialized(ContextLoaderListener.java:103) ~[spring-web-5.1.4.RELEASE.jar:5.1.4.RELEASE]
	at org.eclipse.jetty.server.handler.ContextHandler.callContextInitialized(ContextHandler.java:957) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.servlet.ServletContextHandler.callContextInitialized(ServletContextHandler.java:553) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.handler.ContextHandler.startContext(ContextHandler.java:922) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.servlet.ServletContextHandler.startContext(ServletContextHandler.java:365) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.webapp.WebAppContext.startWebapp(WebAppContext.java:1497) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.webapp.WebAppContext.startContext(WebAppContext.java:1459) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.handler.ContextHandler.doStart(ContextHandler.java:852) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.servlet.ServletContextHandler.doStart(ServletContextHandler.java:278) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.webapp.WebAppContext.doStart(WebAppContext.java:545) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:108) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.doStart(GzipHandler.java:406) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:117) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.Server.start(Server.java:415) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:108) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.server.Server.doStart(Server.java:382) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
	at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:144) ~[steve.jar:?]
	at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:37) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.start(Application.java:57) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.main(Application.java:46) ~[steve.jar:?]
```

goekay, 2019-05-22:
> Mysql and Java are running on the same machine, so I can't figure out how can they have different time zones.

since UTC is suggested by OCPP, steve explicitly sets the time zone to UTC, regardless of what the default of the system is. you can set the time zone to UTC in mysql with the following:

```
SET GLOBAL time_zone = '+00:00';
```

victormunoz, 2019-05-22:
Thank you. It worked!
Maybe this line should appear in the migration sqls...

closed 2019-05-22 by victormunoz

goekay, 2019-05-22:
> Maybe this line should appear in the migration sqls...

running this command applies only to the _live instance_. if you restart mysql, it will revert back to the default. therefore, for the config change to be persistent, the option in config file should be set. documentation: https://dev.mysql.com/doc/refman/5.7/en/time-zone-support.html#time-zone-variables

as you can imagine, this is a hardcore administrative change, since it affects all databases. therefore, it must be done with precaution.