steve-community/steve #140: Maven /JDK version

BobOntario, 2019-05-30:
Hello all

I am struggling to build the project. always got the following error.  

I tried OpenJDK 11.02 + Maven 3.6.0  
Oracle JDK 11.02 + Maven 3.6.0, 3.6.1 and 3.5.4 

Any advice is highly appreciated.

WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
Log file: /root/logs/steve.log
Starting..WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.springframework.cglib.core.ReflectUtils$1 (file:/root/steve-steve-3.2.0/target/steve.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of org.springframework.cglib.core.ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

goekay, 2019-05-30:
these are just warnings. they do not affect the build or the functionality. please see issue #127

BobOntario, 2019-05-30:
thanks.  BTW, after I changed server IP from 127.0.0.1 to my server IP, always got an exception as below, any ideas:

[ERROR] 2019-05-31 03:21:09,975 de.rwth.idsg.steve.SteveProdStarter - Exception happened
java.net.BindException: Cannot assign requested address
	at sun.nio.ch.Net.bind0(Native Method) ~[?:?]
	at sun.nio.ch.Net.bind(Net.java:461) ~[?:?]
	at sun.nio.ch.Net.bind(Net.java:453) ~[?:?]
	at sun.nio.ch.ServerSocketChannelImpl.bind(ServerSocketChannelImpl.java:227) ~[?:?]
	at sun.nio.ch.ServerSocketAdaptor.bind(ServerSocketAdaptor.java:80) ~[?:?]
	at org.eclipse.jetty.server.ServerConnector.openAcceptChannel(ServerConnector.java:331) ~[steve.jar:?]
	at org.eclipse.jetty.server.ServerConnector.open(ServerConnector.java:299) ~[steve.jar:?]
	at org.eclipse.jetty.server.AbstractNetworkConnector.doStart(AbstractNetworkConnector.java:80) ~[steve.jar:?]
	at org.eclipse.jetty.server.ServerConnector.doStart(ServerConnector.java:235) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at org.eclipse.jetty.server.Server.doStart(Server.java:398) ~[steve.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve.jar:?]
	at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:144) ~[steve.jar:?]
	at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:37) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.start(Application.java:50) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.main(Application.java:44) ~[steve.jar:?]

goekay, 2019-05-30:
not sure, but probably the ip is not available. the server might have a different ip than the one you think it has. please google the exception with its message for further help.

closed 2019-05-30 by goekay