steve-community/steve #141: concurrentTX  starttransaction

Gior80, 2019-06-10:
hello, 

i noticed strange behaviour when calling starttransaction.  it always return concurrentTX and i cannot realize why.   Looking at calls in code i see  you  made a check to see if there's  open transaction on the selected chargebox /connector by detecting a transaction with null  stop time and value

getChargeBoxIdsOfActiveTransactions(idTag)

i think this is completely right logic iteself,  what i i think is  wrong is that this check , when calling  starttransaction is made after the new transaction has been inserted to database.  
This way it always return a cuncurrent tx as the jsut inserted transaction has stop value and stop time == null

i think the right flow on startTransction would be first to check for concurrentTX and after insert the transaction to db

goekay, 2019-06-10:
> getChargeBoxIdsOfActiveTransactions(idTag)

there are more steps after this call, though, please look at [this](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/OcppTagServiceImpl.java#L126-L136).
basically, we are checking from which station this request comes from and decide the response accordingly. this was discussed in #73. are you using an older version of steve?

Gior80, 2019-06-10:
hello @goekay i was thinking not,  but lookign at code then yes, it has changed.  

i think we can close that !

thanks

closed 2019-06-10 by goekay