steve-community/steve #142: Add Docker support

aakobordenau, 2019-06-11:
Hi, is there any interest in adding docker support to steve?

goekay, 2019-06-12:
yes. there is even an initial version of dockerized steve. but since i am not an expert in this area, i do not know the best practices and therefore i was hesitant to check it in. because of the same reasons, i do not think that i can maintain the docker version _at the moment_. i simply need time to delve into the world of docker which i lack right now.

aakobordenau, 2019-06-18:
Nice! Can I take a look at the draft anywhere?
I started to dockerize steve as well, but I may miss something because I didn't really dive deep enough into the project yet.

goekay, 2019-06-19:
sure. actually, my colleague @muued worked on "dockerization" of steve. and his changes can be found in the docker branch of [his fork](https://github.com/muued/steve). this is the relevant commit: https://github.com/muued/steve/commit/6211b9ef620cb961efd353041eb3463aa4ed2b66

lingster, 2019-08-14:
@aakobordenau @goekay I've just submitted a PR for this request. Could someone kindly review and provide feedback? You should be able to start Steve out of the box now!

closed 2019-08-19 by goekay