steve-community/steve #145: Testing Mennekes and J1772 type with Steve : Issues

abdulgig, 2019-06-16:
We are trying to test with Mennekes and J1772 type. 

It is from https://oc1.oncharger.com/

It gets connected but when starting a transaction, we received the following error, an error with Temperature format, 

```
[4,"137","FormationViolation","The payload for action could not be deserialized",{"errorMsg":"Cannot deserialize value of type `ocpp.cs._2015._10.UnitOfMeasure` from String \"Celcius\": value not ..."}]

```
JSON request from ChargePoint,

```
[2,"137","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2019-06-14T10:21:29Z","sampledValue":[{"value":"0.00","measurand":"Current.Import","unit":"A"}]},{"timestamp":"2019-06-14T10:21:29Z","sampledValue":[{"value":"0.00","measurand":"Energy.Active.Import.Register","unit":"Wh"}]},{"timestamp":"2019-06-14T10:21:29Z","sampledValue":[{"value":"344","measurand":"Temperature","unit":"Celcius"}]}]}]

```
The ChargePoint Operator says,

```
According to OCPP API you must send "heartbeatInterval" not "interval"
in BootNotification response
[3,"11",{"status":"Accepted","currentTime":"2019-06-14T10:26:46.006Z","interval":7200}] -> you have to change "interval" to "heartbeatInterval"

```

How can the above can be fixed,

1. We need to send heartbeatInterval in BootNotification response
2. Temperature should be in Celsius

Best Regards,

abdulgig, 2019-06-16:
Please bear with me but from where I can control these parameters ?

goRaspy, 2019-06-16:
There is several generations of mennekes charging stations. But for my
understanding they all need an external controler to support occp.

abdulgig, 2019-06-16:
There is no way we could change what we send to the ChargePoint ?

goekay, 2019-06-16:
> We need to send heartbeatInterval in BootNotification response

`heartbeatInterval` can be changed globally for all stations in Settings page from the menu.

> Temperature should be in Celsius

yeah. this is a bug of the charging station software. i will not change steve to compensate a typo of developers. such a mistake is really ridiculous.

abdulgig, 2019-06-16:
```According to OCPP API you must send "heartbeatInterval" not "interval"
in BootNotification response```

What was meant here is not the value, rather it is the parameter name itself. There are 2 parameters,

1. interval
2. heartbeatinterval

As per the ChargingStaing, they need us to send heartbeatinterval parameter rather than "interval"

For the issue with temperature, can you point me to the file in the "src", we may look into it.

Last but not the least, is there a documentation available on how to use Steve, for example,

1. Adding Your First Station
2. How to Add RFID cards
3. How to Bill

This is great work, some more HowTos will drastically improve it. I want to contribute.

Best Regards,

goekay, 2019-06-16:
> What was meant here is not the value, rather it is the parameter name itself. There are 2 parameters

you, as a user, should not care about the name of the parameter. steve supports multiple ocpp versions: 1.2, 1.5 and 1.6. the developers of the protocol changed the name of the parameter from `heartbeatinterval` to `interval` in 1.6. steve internally has the necessary converters and mappings for correct behaviour for specific versions. i think you are making a configuration mistake (with steve or station) such that steve thinks that the station uses another version and therefore uses a wrong processing pipeline.

> For the issue with temperature, can you point me to the file in the "src", we may look into it.

you can find a workaround for a similar problem [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/custom/MeterValue15Deserializer.java). warning: to do something similar requires serious know-how of our toolset and libraries.

> Last but not the least, is there a documentation available on how to use Steve, ...

no there is not, but the tasks themselves are just filling data in forms and saving in database. they can be  under the menu Data Management.

goekay, 2019-06-16:
regarding heartbeatInterval vs interval:

since station developers made the mistake of using Celcius instead of Celsius, i can imagine that they might have made a similar mistake with the parameter name in BootNotificationResponse.

csamsel, 2019-06-16:
> We are trying to test with Mennekes and J1772 type.
> 
> It is from https://oc1.oncharger.com/

So the charging station is not actually made by Menneskes but you are refering to the Mennkes (Typ 2) plug i guess. Feel free to report the bugs in their software to oncharger.

goekay, 2019-06-16:
off topic @abdulgig: i see you copied steve (not a github fork!) and changed its name to give the impression that `enertia-ocpp` is your own product. not cool.

abdulgig, 2019-06-16:
@goekay , thanks for the responses above.  Steve is opensource, therefore, its pointless to debate on anything across those lines.

abdulgig, 2019-06-16:
> you, as a user, should not care about the name of the parameter. steve supports multiple ocpp versions: 1.2, 1.5 and 1.6. the developers of the protocol changed the name of the parameter from [...]

I will try to add their charger under 1.5 and 1.2 to test. 

> > For the issue with temperature, can you point me to the file in the "src", we may look into it.
> 
> you can find a workaround for a similar problem [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/custom/MeterValue15Deserializer.java). warning: to do something similar requires serious know-how of our toolset and libraries.
> 
> > Last but not the least, is there a documentation available on how to use Steve, ...
> 
> no there is not, but the tasks themselves are just filling data in forms and saving in database. they can be found under the menu Data Management.

abdulgig, 2019-06-16:
This is what oc1 guys say, they claim that we are doing it wrong, anyways, need to find a middleway

[image]

goekay, 2019-06-16:
> Steve is opensource, therefore, its pointless to debate on anything across those lines.

having the source code open does not automatically give you the rights to do anything you wish with the project, therefore, we additionally have licenses and copyrights. your project does not credit steve (and its contributors) which it bluntly copied and renamed. you are being a bad open source citizen, but also are violating copyrights. please educate yourself.

moreover, you are expecting our help to fix your problems (which are not steve's problems). you are being disrespectful and i will not help you anymore.

closed 2019-06-16 by goekay

abdulgig, 2019-06-16:
@goekay , if you don't have the courage to share, then please stop advertising or promoting opensource. If it's a matter of copying then Windows is the biggest.

I have seen your responses elsewhere and I have found them to be pretty rude. Educate yourself as in how to communicate with people before you even think of development.

People who usually point fingers at others can't face when one is pointed at them.

Don't run an opensource forum when you want to act God on everything. Respect first before you expect!

csamsel, 2019-06-16:
> This is what oc1 guys say, they claim that we are doing it wrong, anyways, need to find a middleway
> 
> [image]

Yes Degree Celsius is fine both for OCPP and SteVe.

But they don't know how to spell the word Celsius.

goekay, 2019-06-16:
> if you don't have the courage to share, then please stop advertising or promoting opensource.

what does this have anything to do with _courage to share_? sorry, but you clearly don't know how open source, licensing and copyrights work. therefore, it was just a factual statement, when i said "educate yourself".

> If it's a matter of copying then Windows is the biggest.

what? what does this have anything to do with our discussion? you are still missing the point. it's not about copying. in fact, GPLv3 allows copying and modifying. my problem was about _crediting steve and respecting copyrights_.

> I have seen your responses elsewhere and I have found them to be pretty rude. Educate yourself as in how to communicate with people before you even think of development.
>
> People who usually point [...]

now... i could give counter-arguments, but this is getting personal. don't do this. i attribute this tone of yours to being wrong and being caught doing something you shouldn't have done.

by the way, thanks for deleting your repo.

abdulgig, 2019-06-16:
> what does this have anything to do with _courage to share_? sorry, but you clearly don't know how open source, licensing and copyrights work. therefore, it was just a factual statement, when i said [...]

Please correct yourself, I haven't deleted the repo and neither I would , not there is a need to ! Its private as I intend to develop on top of it and I'll make it public once I finish.  It would be free to use and I won't lecture on ethics when its  not needed.

Thanks, but no thanks !

abdulgig, 2019-06-16:
> 
> 
> > This is what oc1 guys say, they claim that we are doing it wrong, anyways, need to find a middleway
> > [image]
> 
> Yes Degree Celsius is fine both for OCPP and SteVe.
> 
> But they don't know how to spell the word Celsius.

I think that can be addressed with some modification.

For the "heartbeatinterval" and "interval", I am confused it changed in 1.6 and was available in 1.2 , 1.5.

Still trying to do some research there

goekay, 2019-06-16:
>Please correct yourself, I haven't deleted the repo and neither I would , not there is a need to ! Its private as I intend to develop on top of it and I'll make it public once I finish. It would be free to use and I won't lecture on ethics when its not needed.
>
>Thanks, but no thanks !

well, it was public before you changed it to private (after my comment), so it appears as "not found" to everyone of us. thanks for the correction! :wink:

i find it great that you continue to develop it. before you make it public again, just do not forget again to respect the license, copyrights and credit steve. to make things easier for you, today i even added the copyright notice to each file. so your files should contain this exact header in each file. 

steve is free to use, too! a friendly reminder of what you can do with steve: https://tldrlegal.com/license/gnu-general-public-license-v3-(gpl-3)

you see how friendly i am? i still cannot comprehend that you thought i am rude and do not respect others. :unamused:

/s

abdulgig, 2019-06-16:
Good to see you getting better with communication ! That is nice  :wink:

Now if you say a word after this, I am going to make the repository public again and will start appearing in search results  :sunglasses:

I have a few more companies lined up and I am going to test your steVe with it. Lets see! With the OnCharge guys, we'll do the changes with them and share the results.

By the way, nice meeting you :grin:

lnjuanj, 2019-06-17:
Wow, this is pretty bad, Abdul,
I just hope it is easy to catch you if you don’t give credits to the
original repo.
Someone sharing means you should be respectful with their work. In this
case, hard work that has helped many people develop their charging points,
including myself. BTW, CelCius is just... ridiculous 😂 “we support CelCius
but not Celsius” 🤣 😔

abdulgig, 2019-06-17:
I think there should be a way where this parameter setting should be controllable by user, atleast(Advanced User) for better integration with others.  

Secondly, it isn't about what we support or not, the point was how flexible could SteVe be to accommodate / handle such cases. May be an embedded tool, similar to OCA's compliance testing !

goekay, 2019-06-17:
> I think there should be a way where this parameter setting should be controllable by user, atleast(Advanced User) for better integration with others.
>
> Secondly, it isn't about what we support or [...]

yeah.. no. 1) advanced users can code and change the source code. so, there is always that possibility. 2) "for better integration with others" developers can just stop programming errors and spelling mistakes in parameter names. the protocols and interfaces exists for better integration with others and are the contract between two parties. both parties should start doing the right thing. 3) steve does not have to be flexible enough to handle spelling mistakes of other programmers.

instead of complaining here, go and complain to the developers of the station.

abdulgig, 2019-06-17:
@goekay , and here we go again,  well please stop being so emotional about it ! I am NOT complaining.

SteVe does not have to be this , SteVe does not have be that. For that matter, make it proprietary and stop snubbing people if they share their viewpoint in the community.

We sorted out the parameters with OnCharger and seems to be working now. So that is over !

eviontech, 2019-12-17:
Can i change the package name to remove steve name so it doesnt appear in logs ? is it valid under the license

josecharry, 2020-09-01:
I have tested steve with J1772 and it works very good. I didn't have to make any change in the source code