steve-community/steve #147: error after mvn package

Ivra23, 2019-06-20:
Hello, 
I have this problem but i don't know what's this.
[image]

Version of my Java and Maven 
[image]

Version of my WampServer
[image]

Thank you

goekay, 2019-06-20:
you should google the error message of the first screenshot. explanation: https://stackoverflow.com/a/45013815

Ivra23, 2019-06-28:
Hello,
Thank you for your help,
now i have antahor problem with the version 3.3.0 

[image]

I put the correct time zone normally and give all guranted to the user steve
[image]

Thank you

goekay, 2019-06-28:
the database exception you see has nothing to do with the time zone. the exception message is clear: the user `steve` does not have the necessary permissions to access the database. please make sure that you executed the database preparation steps from readme.

Ivra23, 2019-06-28:
Yes I followed step by step readme, but maybe the version of the mariadb it's not correct no ?
[image]

goekay, 2019-06-28:
i want to remind you the following: github issues are not a support channel.

maybe first try to connect as steve instead of root, to see if it works at all. and you are using mariadb version 10.1.23 even though in readme point out that 10.2.1 or later works.

Ivra23, 2019-06-28:
oki thank you i will try.

closed 2019-06-28 by goekay