steve-community/steve #148: Allow empty custom configuration values

lategoodbye, 2019-06-21:
#### Checklist

[X] I checked other issues already and found no answer
[X] I checked the documentation and found no answer
[X] I am running the version 3.2.0 and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

I want to change a custom configuration value to an empty string. Unfortunately the SteVe form for OCPP 1.6 change configuration always enforces a non-empty string ("Value is required").

#### Describe the solution you'd like

It would be nice to allow empty custom configuration values, because it up to the OCPP client to do this validation.

goekay, 2019-06-21:
i can see the use of this feature request. however, ocpp spec does not specify such a gray area, does it? moreover, the default behaviour of xml message serialization trims the empty `value` tag (which per ocpp spec is required) and sends only the `key` in message.

example with value:
`<changeConfigurationRequest xmlns="urn://Ocpp/Cp/2015/10/"><key>ConnectionTimeOut</key><value>2</value></changeConfigurationRequest>`

example with empty value:
`<changeConfigurationRequest xmlns="urn://Ocpp/Cp/2015/10/"><key>ConnectionTimeOut</key></changeConfigurationRequest>`

overriding this behaviour is possible, but it is there for a reason. i am just afraid, that we would be opening a can of worms, when we implement this. any ideas?

lategoodbye, 2019-06-21:
Okay, my fault i was only focusing on JSON, which permits an empty value. So i understand you don't want to handle them differently. Unfortunately i don't have a solution for the SOAP case.

goekay, 2019-06-21:
> Okay, my fault i was only focusing on JSON, which permits an empty value. 

can you point me to the part in the spec where this is defined?

> So i understand you don't want to handle them differently. Unfortunately i don't have a solution for the SOAP case.

no i don't. both should have the same behaviour/implementation. SOAP was just an example. we can change this.

actually, i am looking at this from a higher level perspective: does ocpp spec allows such empty values? i couldn't see any signs in this regard. 

the dicussion is similar to existence of `NULL` in programming languages :)

lategoodbye, 2019-06-21:
I put the JSON schema of ChangeConfiguration.json into a Schema validator and it accepts the following (doesn't make sense):

`
{"key":"ConnectionTimeout","value":""}
`

I agree, i couldn't find a place which permits empty values explicitly. But i also couldn't find a place in the spec which prohibits empty values.

goekay, 2019-06-21:
> I agree, i couldn't find a place which permits empty values explicitly. But i also couldn't find a place in the spec which prohibits empty values.

yes. because of this i called this a _gray area_. i will look into it. maybe i can find a happy medium.

goekay, 2019-06-21:
by the way, maybe you and other charging station users can give me examples where the charging station defines or permits empty value. i need some real world use cases.

lategoodbye, 2019-06-21:
Sure. For custom fields i can think of a GPRS connection it might be necessary to leave the APN password empty. Another idea would be the EVSE ID as an optional parameter.

goekay, 2019-06-21:
okay, the implementation was easier than i thought, because i didn't have to make big changes that would affect other aspects of our processing pipeline. i will commit it in a few minutes.

lategoodbye, 2019-06-21:
I think applying this to all would be more consistent. But i'm not aware of any side effects.

commit 547b8776 (2019-06-21): permit empty values for conf. keys (Closes #148)
closed 2019-06-21 by goekay