steve-community/steve #149: WARNING: An illegal reflective access operation has occurred

pSnehanshu, 2019-06-21:
#### Specifications
```
SteVe Version     : 3.3.0
Operating system  : Ubuntu 18.04.2
JDK               : openjdk version "11.0.3"
Database          : mysql  Ver 14.14 Distrib 5.7.26, for Linux (x86_64) using  EditLine wrapper
```
The following command is successful.
```
mvn package
```
But 
```
java -jar target/steve.jar
```
produces an error.
<pre>Log file: /home/sp/logs/steve.log
Starting......WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.jooq.tools.reflect.Reflect (file:/home/sp/app/steve-steve-3.3.0/target/libs/jooq-3.11.11.jar) to constructor java.lang.invoke.MethodHandles$Lookup(java.lang.Class)
WARNING: Please consider reporting this to the maintainers of org.jooq.tools.reflect.Reflect
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
........... FAILED!
Please refer to the log file for details
</pre>

The contents of the log file:
<pre>[INFO ] 2019-06-21 19:52:48,745 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property &apos;main.properties&apos; can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2019-06-21 19:52:48,917 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the &apos;PROD&apos; profile
[INFO ] 2019-06-21 19:52:48,944 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2019-06-21T19:52:48.927Z
[INFO ] 2019-06-21 19:52:48,996 org.eclipse.jetty.util.log - Logging initialized @1668ms to org.eclipse.jetty.util.log.Slf4jLog
[INFO ] 2019-06-21 19:52:49,394 org.eclipse.jetty.server.Server - jetty-9.4.17.v20190418; built: 2019-04-18T19:45:35.259Z; git: aa1c656c315c011c01e7b21aabb04066635b9f67; jvm 11.0.3+7-Ubuntu-1ubuntu218.04.1
[INFO ] 2019-06-21 19:52:49,649 org.eclipse.jetty.server.session - DefaultSessionIdManager workerName=node0
[INFO ] 2019-06-21 19:52:49,649 org.eclipse.jetty.server.session - No SessionScavenger set, using defaults
[INFO ] 2019-06-21 19:52:49,650 org.eclipse.jetty.server.session - node0 Scavenging every 660000ms
[INFO ] 2019-06-21 19:52:49,662 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2019-06-21 19:52:49,663 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2019-06-21 19:52:51,865 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
[INFO ] 2019-06-21 19:52:52,165 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
[INFO ] 2019-06-21 19:52:55,732 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2019-06-21 19:52:56,544 org.apache.cxf.endpoint.ServerImpl - Setting the server&apos;s publish address to be /CentralSystemServiceOCPP12
[INFO ] 2019-06-21 19:52:56,565 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2019-06-21 19:52:56,687 org.apache.cxf.endpoint.ServerImpl - Setting the server&apos;s publish address to be /CentralSystemServiceOCPP15
[INFO ] 2019-06-21 19:52:56,688 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
[INFO ] 2019-06-21 19:52:56,800 org.apache.cxf.endpoint.ServerImpl - Setting the server&apos;s publish address to be /CentralSystemServiceOCPP16
[INFO ] 2019-06-21 19:52:56,802 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2019-06-21 19:52:56,831 org.apache.cxf.endpoint.ServerImpl - Setting the server&apos;s publish address to be /CentralSystemService
[INFO ] 2019-06-21 19:52:58,103 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: Ant [pattern=&apos;/static/**&apos;], []
[INFO ] 2019-06-21 19:52:58,103 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: Ant [pattern=&apos;/views/**&apos;], []
[INFO ] 2019-06-21 19:52:58,160 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: any request, [org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter@4d529dbf, org.springframework.security.web.context.SecurityContextPersistenceFilter@5c59dfc7, org.springframework.security.web.header.HeaderWriterFilter@73a88711, org.springframework.security.web.csrf.CsrfFilter@267a8777, org.springframework.security.web.authentication.logout.LogoutFilter@4960b819, org.springframework.security.web.authentication.UsernamePasswordAuthenticationFilter@3ba015b1, org.springframework.security.web.authentication.www.BasicAuthenticationFilter@5b58f639, org.springframework.security.web.savedrequest.RequestCacheAwareFilter@2ef63b6d, org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter@535aa7c7, org.springframework.security.web.authentication.AnonymousAuthenticationFilter@76641439, org.springframework.security.web.session.SessionManagementFilter@43f5b014, org.springframework.security.web.access.ExceptionTranslationFilter@d62feaf, org.springframework.security.web.access.intercept.FilterSecurityInterceptor@3df4cd96]
[INFO ] 2019-06-21 19:52:58,510 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.0.16.Final
[INFO ] 2019-06-21 19:52:59,139 org.springframework.web.context.ContextLoader - Root WebApplicationContext initialized in 9476 ms
[INFO ] 2019-06-21 19:52:59,233 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring DispatcherServlet &apos;spring-dispatcher&apos;
[INFO ] 2019-06-21 19:52:59,233 org.springframework.web.servlet.DispatcherServlet - Initializing Servlet &apos;spring-dispatcher&apos;
[INFO ] 2019-06-21 19:52:59,242 org.springframework.web.servlet.DispatcherServlet - Completed initialization in 9 ms
[INFO ] 2019-06-21 19:52:59,262 org.eclipse.jetty.server.handler.ContextHandler - Started o.e.j.w.WebAppContext@406cc81b{/steve,jar:file:/home/sp/app/steve/target/steve.jar!/webapp,AVAILABLE}
[ERROR] 2019-06-21 19:52:59,478 de.rwth.idsg.steve.SteveProdStarter - Exception happened
java.io.IOException: Failed to bind to steve.snehanshu.now.sh/35.181.83.190:8080
	at org.eclipse.jetty.server.ServerConnector.openAcceptChannel(ServerConnector.java:346) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.ServerConnector.open(ServerConnector.java:308) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.AbstractNetworkConnector.doStart(AbstractNetworkConnector.java:80) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.ServerConnector.doStart(ServerConnector.java:236) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.Server.doStart(Server.java:396) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
	at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:162) ~[steve.jar:?]
	at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:55) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.start(Application.java:75) ~[steve.jar:?]
	at de.rwth.idsg.steve.Application.main(Application.java:64) ~[steve.jar:?]
Caused by: java.net.BindException: Cannot assign requested address
[... 104 lines cut ...]
	at org.eclipse.jetty.server.AbstractNetworkConnector.doStart(AbstractNetworkConnector.java:80) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.ServerConnector.doStart(ServerConnector.java:236) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
[... cut ...]
[INFO ] 2019-06-21 20:07:07,032 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown initiated...
[INFO ] 2019-06-21 20:07:07,065 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown completed.
[INFO ] 2019-06-21 20:07:07,068 org.eclipse.jetty.server.handler.ContextHandler.steve - Closing Spring root WebApplicationContext
[INFO ] 2019-06-21 20:07:07,069 org.eclipse.jetty.server.handler.ContextHandler - Stopped o.e.j.w.WebAppContext@264340f0{/steve,jar:file:/home/sp/app/steve-steve-3.3.0/target/steve.jar!/webapp,UNAVAILABLE}
</pre>

goekay, 2019-06-21:
did you check other issues? this is already covered by previous issues: https://github.com/RWTH-i5-IDSG/steve/issues/54#issuecomment-380651115, https://github.com/RWTH-i5-IDSG/steve/issues/137

closed 2019-06-21 by goekay

pSnehanshu, 2019-06-21:
Actually, I fixed the problem partly. 
I am trying to run SteVe on a Azure VPS. The VPS has a private IP and a public IP.

First I set `server host` to 127.0.0.1, it worked. But only from the same machine. It wasn't accessible from the outside world. Then I set it to 0.0.0.0, SteVe now binds to the private ip 10.0.x.x, but it's still isn't accessible from outside world. Setting `server host` to public address 20.188.x.x fails when running `java -jar target/steve.jar`.
Please help me. How can I make SteVe accessible from the internet?

goekay, 2019-06-22:
127.0.0.1 is there only for local access. 0.0.0.0 listens to [all network interfaces](https://github.com/RWTH-i5-IDSG/steve/wiki/Configuration#multiple-interfaces). so, actually using 0.0.0.0 should have solved your problem. why it didn't, i cannot answer. probably your azure vps configuration needs some tweaking. but with this, we are stepping outside the world steve. i cannot help you there, sorry.