steve-community/steve #150: Steve sends a FIN ACK, closing the conexion, after synchronizing and connecting to client

fusionJose, 2019-06-25:
Hi,

I've made a client project to connect with steve. When I connect to it through a websocket, steve closes the connection sending a [FIN, ACK]. Is there any way to see some log? or any idea?

Here is what i see through Wireshark: (steve is on port 8080)

[image]

goekay, 2019-06-25:
Can you please describe what the problem was and how you solved it. It might be useful for people with a similar problem.

fusionJose, 2019-06-26:
I noticed that my client was sending in the header of the HTTP frame a Connection: close, upgrade. And Steve blows up with it. So I changed and configured my client to send just Connection: upgrade. Steve works only with that header.

Is there any standard for that?, I understand that a Connection: close, upgrade is not bad.

goekay, 2019-06-26:
> Is there any standard for that?

of course. 

you should not write a websocket implementation without following the protocol spec as defined in [RFC-6455](https://tools.ietf.org/html/rfc6455) by Internet Engineering Task Force (IETF). more specifically, we are relying on the [websocket implementation](https://www.eclipse.org/jetty/documentation/current/websocket-intro.html#ws-intro-provides) of our server library Jetty. i cannot envision that the spec is wrongly implemented by the developers of Jetty. i would advise to use a battle-tested websocket library in your programming language.

closed 2019-06-26 by goekay