steve-community/steve #153: Jooq excute jar

Ivra23, 2019-07-01:
Hello,
You can help me because I find the solution to my problem but that doesn't resolve itself, 
I found that it was a problem of logo but in the file log4j2.xml tthis line 
fix this problem 
**https://github.com/jOOQ/jOOQ/ issues / 4019
<AsyncLogger name = "org.jooq.Constants" level = "WARN" />**

[image]

But in excute programm it persists.

Thank you

goekay, 2019-07-01:
sorry but i don't understand what the problem is. here are a couple of things:
- jooq logo is not a problem. i don't know why you bring it up.
- the warnings from the screenshot are not a problem. they do not cause the fail.
- as the last log line mentions, please take a look at the log file and find the exception for the actual reason of the problem.

Ivra23, 2019-07-01:
okeay sorry for the inconvenience, 
I will delete my issues :)

closed 2019-07-01 by goekay