steve-community/steve #157: Migration scripts break when database was configured with Default Engine = MyISAM

csamsel, 2019-07-12:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :  3.3.0
Operating system  : not relevant
JDK               : not relevant
Database          : MariaDB 5.5.57
```

#### Expected Behavior

SteVe should create all tables explicitly with InnoDB. Migration scripts should work

#### Actual Behavior

Migration script fails with: 
```
[INFO] Scanning for projects...
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Building steve 3.3.0
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M2:enforce (enforce-java) @ steve ---
[INFO] 
[INFO] --- git-commit-id-plugin:2.2.4:revision (default) @ steve ---
[INFO] 
[INFO] --- properties-maven-plugin:1.0.0:read-project-properties (default) @ steve ---
[INFO] 
[INFO] --- flyway-maven-plugin:5.2.4:migrate (default) @ steve ---
[INFO] Flyway Community Edition 5.2.4 by Boxfuse
[INFO] Database: jdbc:mysql://localhost:3306/stevedb (MySQL 5.5)
[INFO] Successfully validated 31 migrations (execution time 00:00.208s)
[INFO] Creating Schema History table: `stevedb`.`schema_version`
[INFO] Current version of schema `stevedb`: << Empty Schema >>
[WARNING] outOfOrder mode is active. Migration of schema `stevedb` may not be reproducible.
[INFO] Migrating schema `stevedb` to version 0.6.6 - inital
[INFO] Migrating schema `stevedb` to version 0.6.7 - update
[INFO] Migrating schema `stevedb` to version 0.6.8 - update
[INFO] Migrating schema `stevedb` to version 0.6.9 - update
[INFO] Migrating schema `stevedb` to version 0.7.0 - update
[INFO] Migrating schema `stevedb` to version 0.7.1 - update
[INFO] Migrating schema `stevedb` to version 0.7.2 - update
[INFO] Migrating schema `stevedb` to version 0.7.3 - update
[INFO] Migrating schema `stevedb` to version 0.7.6 - update
[INFO] Migrating schema `stevedb` to version 0.7.7 - update
[INFO] Migrating schema `stevedb` to version 0.7.8 - update
[INFO] Migrating schema `stevedb` to version 0.7.9 - update
[INFO] Migrating schema `stevedb` to version 0.8.0 - update
[INFO] Migrating schema `stevedb` to version 0.8.1 - update
[INFO] Migrating schema `stevedb` to version 0.8.2 - update
[INFO] Migrating schema `stevedb` to version 0.8.4 - update
[INFO] Migrating schema `stevedb` to version 0.8.5 - update
[WARNING] DB: Alter  table `stevedb`.`charge_box` with foreign key constraint failed. Referenced table `stevedb`.`address` not found in the data dictionary near '
FOREIGN KEY (address_pk) REFERENCES address (address_pk) ON DELETE SET NULL ON UPDATE NO ACTION'. (SQL State: HY000 - Error Code: 150)
[WARNING] DB: Can't create table 'stevedb.#sql-4536_7c0f5' (errno: 150) (SQL State: HY000 - Error Code: 1005)
[ERROR] Migration of schema `stevedb` to version 0.8.5 - update failed! Please restore backups and roll back database and code! 
```

#### Steps to Reproduce the Problem

  1. Configure MariaDB / MySQL with default engine MyISAM
  2. Install StVe
  3. 

#### Additional context

I think we have to go through all migration scripts and change them to force InnoDB for table creation. I thought we could just set the default engine during database creation and add that to the docu but it seems that's not possible.

goekay, 2019-07-12:
thanks for the detailed report. i tried to reproduce the error but could not. i used the same version for windows x64 downloaded from [here](https://downloads.mariadb.org/mariadb/5.5.57/). i started the db without any custom configuration file. i could see log entries regarding InnoDB setup, so i guess the default is InnoDB already?

moreover, database migration scripts were successful until `0.9.7` which failed because "_View's SELECT contains a subquery in the FROM clause_". this is the reason why i bumped the database version at the time so the failure is justified, since this version does not support it. so even if you overcame the problem you illustrated, you would have encountered this one with a subsequent script.

on the other hand...

> I think we have to go through all migration scripts and change them to force InnoDB for table creation.

if we would do that, an upgrade to a newer steve version would be a hassle due to checksum checks. we would force users to do an exra step, which we did once [before](https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-can-i-upgrade-from-1xx-to-2xx) and i want to prevent this from happening again.

is there a reason you are not using the suggested mysql/mariadb version from readme?

csamsel, 2019-07-12:
I'm trying to reproduce it right now. I'll report back later if i manage to.

| Is there a reason you are not using the suggested mysql/mariadb version from readme?

A colleague wanted to use an existing database cluster. He eventually gave up on that though.

csamsel, 2019-07-12:
I was able to reproduce it in a Ubuntu 18.04 VM with 5.7.26 by setting

`default-storage-engine=MyISAM`

in `/etc/mysql/mysql.conf.d/mysqld.cnf`

removing it makes the error go away.

What also works is (with mysql having MyISAM as default storage engine):

`FLYWAY_INIT_SQL="SET default_storage_engine=InnoDB;" mvn package`

Havent found how to get the INIT_SQL variable into the `pom.xml` though. Any idea?

closed 2019-07-13 by goekay

csamsel, 2019-07-13:
Jup that works. Thanks!

commit c3691d78 (2019-07-13): flyway: explicitly set default engine to InnoDB (Closes #157)