steve-community/steve #158: Build failure when updating to Jun 28, 2019 commits

TechOverflow, 2019-07-16:
#### Checklist

- [ x ] I checked other issues already, but found no answer/solution
- [ x ] I checked the documentation and wiki, but found no answer/solution
- [ x ] I am running the latest version and the issue still occurs
- [ x ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.3.0
Operating system  : Ubuntu 16.04
JDK               : 11
Database          : MySQL
```

#### Expected Behavior

Build success

#### Actual Behavior

```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.0:compile (default-compile) on project steve: Compilation failure: Compilation failure: 
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/service/CentralSystemService16_Service.java:[28,27] cannot find symbol
[ERROR]   symbol:   class TransactionStopEventActor
[ERROR]   location: package jooq.steve.db.enums
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[31,27] cannot find symbol
[ERROR]   symbol:   class TransactionStopEventActor
[ERROR]   location: package jooq.steve.db.enums
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[56,35] cannot find symbol
[ERROR]   symbol:   class TransactionStart
[ERROR]   location: package jooq.steve.db.tables
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[56,1] static import only from classes and interfaces
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[57,35] cannot find symbol
[ERROR]   symbol:   class TransactionStop
[ERROR]   location: package jooq.steve.db.tables
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[57,1] static import only from classes and interfaces
[ERROR] /opt/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[462,61] cannot find symbol
[ERROR]   symbol:   class TransactionStopEventActor
[ERROR]   location: class de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl
```

#### Steps to Reproduce the Problem

  1. Followed changes in commit 
https://github.com/RWTH-i5-IDSG/steve/commit/cfe2fde02acbc9b5e993f0811e31d5a84b59b3f1 and https://github.com/RWTH-i5-IDSG/steve/commit/2793b084eaf40f57bae68b532fe63fb2b82707b0

  2. Downloaded steve and replaced entire /src/main/java/de directory, checked pom.xml

#### Additional context

This could be entirely my fault when merging.
I have not tried replacing the entire src as all changes seem to be in /java/.

goekay, 2019-07-17:
some classes related to the database schema seem to be missing. make sure that the migration files under `src/main/resources/db/migration` are present in your repo.

TechOverflow, 2019-07-17:
Thank you! I somehow skipped V0.9.7/8/9.

closed 2019-07-17 by TechOverflow