steve-community/steve #159: Dicussion: should we use github dependabot?

csamsel, 2019-07-23:
https://github.com/marketplace/dependabot-preview

Can automatically make pull requests if it detects security issues in dependencies. Works with maven. Free of charge and it's provided by github themselves.  Sounds good to me.

Any objections?

goekay, 2019-07-25:
sure. why not. if we are not satisfied with it, we can always disable it.

however, i tried using it a while ago after seeing a security issue notification. it failed, because it could not handle version properties/variables which we use. hopefully they improved it.

csamsel, 2019-07-26:
Agreed. I cannot set it up because i'm not an owner of the organization anymore.

muued, 2019-08-22:
should now be set up

closed 2019-08-22 by goekay