steve-community/steve #161: Upgrade to 3.3.1

kernelkraut, 2019-07-26:
This may not be a bug specifically but we are struggling to upgrade from 3.3.0 to 3.3.1 - We have downloaded the new release and run mvn package but mvn keeps building the old version. We are very new to mvn so please forgive the stupid question but how to we upgrade?

#### Specifications

SteVe Version     :  3.3.0
Operating system  : CentOS
JDK               :  11
Database          : MariaDB

goekay, 2019-07-26:
ups. my bad. forgot to bump the version in maven before making the release =/

this does not change the functionality, though. if you download and build the latest commit, it has the bugfix.

kernelkraut, 2019-07-26:
Awesome thanks!

closed 2019-07-26 by kernelkraut

goekay, 2019-07-26:
i fixed the release. now it points to the correct commit with the version change. sorry for the confusion.

kernelkraut, 2019-07-26:
Thanks for the quick help today!