steve-community/steve #162: Eclipse issue building

darknain, 2019-08-12:
Hello,

I downloaded the steve project from git with Eclipse.
I installed jdk 11, maven is already installed with Eclipse and installed MySql.

I did the configuration to build with maven from this post :  [http://www.v2g-evse.com/2018/04/16/ev-charging-station-administration-using-ocpp-1-6-with-steve/](url)

I have one issue in the pom.xml : 
Plugin execution not covered by lifecycle configuration: org.flywaydb:flyway-maven-plugin:5.2.4:migrate (execution: default, phase: generate-sources)

Here is what I get in the console when I run it : 

_> [INFO] Scanning for projects...
> [INFO] 
> [INFO] -------------------------< de.rwth.idsg:steve >-------------------------
> [INFO] Building steve 3.3.2
> [INFO] --------------------------------[ jar ]---------------------------------
> [INFO] 
> [INFO] --- exec-maven-plugin:1.6.0:java (default-cli) @ steve ---
> [WARNING] 
> java.lang.ClassNotFoundException: de.rwth.idsg.steve.Application
> 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:471)
> 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:588)
> 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
> 	at org.codehaus.mojo.exec.ExecJavaMojo$1.run(ExecJavaMojo.java:270)
> 	at java.base/java.lang.Thread.run(Thread.java:834)
> [INFO] ------------------------------------------------------------------------
> [INFO] BUILD FAILURE
> [INFO] ------------------------------------------------------------------------
> [INFO] Total time:  2.853 s
> [INFO] Finished at: 2019-08-12T15:16:01+02:00
> [INFO] ------------------------------------------------------------------------
> [ERROR] Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.6.0:java (default-cli) on project steve: An exception occured while executing the Java class. de.rwth.idsg.steve.Application -> [Help 1]
> [ERROR] 
> [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
> [ERROR] Re-run Maven using the -X switch to enable full debug logging.
> [ERROR] 
> [ERROR] For more information about the errors and possible solutions, please read the following articles:
> [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException_

Best Regards.

goekay, 2019-08-12:
hello, i googled `Plugin execution not covered by lifecycle configuration` and found many issues and workarounds related to eclipse. i switched to intellij idea years ago because of problems like this. maven support in eclipse is buggy.

the console output you've sent indicates that the plugin only tries to run the compiled project, without actually building and compiling first (generate-sources and compile phases etc.). you should maybe follow the steps in our readme first.

goekay, 2019-08-27:
closing due to inactivity.

closed 2019-08-27 by goekay