steve-community/steve #164: Requesting the configuration disconnects the charging station

ameinl, 2019-08-21:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.3.2
Operating system  : Debian Linux - Testing Distribution
JDK               : OpenJDK 11.0.4
Database          : MariaDB 10.3
```

#### Expected Behavior

- I see the configuration of the charging station
- The charging station is still connected to SteVe

#### Actual Behavior

- I do not see the configuration of the charging station
- The charging station is not connected to SteVe anymore

#### Steps to Reproduce the Problem

  1. Boot the charging station; the charging station successfully connects via OCPP 1.6 JSON to SteVe
  2. In SteVe's web portal open the page Operations -> OCPP v1.6 -> Get Configuration
  3. Select the charging station and all parameters
  4. Click on the button "Perform"

#### Additional context

- After having re-booted the charging station, it successfully re-connects to SteVe

goekay, 2019-08-21:
can you please send logs regarding the websocket messages being sent/received and connect/disconnect behaviour?

ameinl, 2019-08-21:
Here you are: [Log.txt](https://github.com/RWTH-i5-IDSG/steve/files/3525379/Log.txt)

goekay, 2019-08-21:
thanks. since other operations are working, i presume that some of the configration keys you select are not supported by this station and it does not know what to do about this (i.e. the following logic is not implemented: send the values for recognized keys and put unrecognized keys in the _unknownKey_ list).

please try sending the Get Configuration message without selecting any key. in this case "unrecognized keys situation" should not occur, because (from spec): _If the list of keys in the request PDU is empty or missing (it is optional), the Charge Point SHALL return a list of all configuration settings in GetConfiguration.conf_

goekay, 2019-08-21:
@ameinl can you please post your findings?