steve-community/steve #193: Credentials for Web interface login / access

jpaquete, 2019-09-06:
Hello I am testing Steve ocpp with good results. However, the credentials for web interface access are not working for the login(shows error Request method 'POST' not supported).
The webinterface browses to /home dashboard without mandatory login... Is that by design ? Or i'm missing some configuration ?... Thanks for a great program !

goekay, 2019-09-07:
>However, the credentials for web interface access are not working for the login(shows error Request method 'POST' not supported).

how are you trying to login? via the web page as a normal user or programmatically making `POST` calls?

> The webinterface browses to /home dashboard without mandatory login... Is that by design ? Or i'm missing some configuration ?

make sure you are running with the `prod` maven profile and not `dev`. in `dev` i disabled login mechanisms to not to have to login each time while developing and testing.

jpaquete, 2019-09-08:
goekay, that was the problem I was using dev :). Many thanks !

closed 2019-09-08 by goekay