steve-community/steve #194: Hot deployment

qwqeq, 2019-09-07:
I wanna to know how to set up Hot deployment while I'm developing.

qwqeq, 2019-09-07:
Why doesn't the page show my changes when I modify the JSP file

goekay, 2019-09-07:
the jsp files have to be compiled to normal java classes for them to be usable. at the moment this is done when building the project. so you have to rebuild steve and restart it.

qwqeq, 2019-09-07:
@goekay But if I do as you said,my development efficiency will be extremely low. If I change one word in a jsp file and then I have to rebuild steve and restart it.

goekay, 2019-09-07:
i agree, it's not optimal. hot reloading/deployment is currently not possible. but i will look into it to improve things for the dev environment.

qwqeq, 2019-09-09:
@goekay I would like to know how you developed Steve. When you modified the JSP file, did you do what you said? Recompile and restart Steve

goekay, 2019-09-09:
yeah, sure. it works but you have to think ahead and have a mental model of the consequences of what you are doing. that eliminates the guesswork and frequent recompiles/restarts.

goekay, 2019-09-09:
by the way, you copied [steve](https://github.com/qwqeq/steve) without any reference to the original repo. moreover, you deleted the contents of the readme & license files and stripped away the copyright notice at the header of each file. not cool. you have to adhere to our license and copyrights.

another recent case was [here](https://github.com/RWTH-i5-IDSG/steve/issues/145#issuecomment-502453196). make sure you read my comments about what you can and cannot do with steve.

[image]

[image]

ijazm123, 2020-02-05:
@goekay Can I package this to war file and deploy in tomcat?

csamsel, 2020-02-05:
No

closed 2020-12-11 by goekay