steve-community/steve #195: Remote strat transaction OCPP ID Tag

jpaquete, 2019-09-11:
Hello, I was able to remote stop a transaction using real hardware but I am not able to do a "Remote Start Transaction" on the ocpp 1.6 operations menu. I cannot get a OCPP ID Tag available on the drop down. What is the purpose on the ocpp tag ? and how can I get it to show up on the dropdown ? (always getting the message "User ID Tag is required". Thanks a lot!

goekay, 2019-09-11:
ocpp id tag (or user id tag) is the identification for a user that interacts with a charging station. in my experience this translates to an rfid card held to a card reader in the charging station to authorize starting and stopping charging sessions. under the data management menu tab, steve has the interface to add/modify/delete id tags. you should first add an id tag which then you can use/reference for remote start transaction. hope this helps.

jpaquete, 2019-09-11:
It helped Tks!. My tags were in blocked state that was why they did not show available.

closed 2019-09-11 by jpaquete