steve-community/steve #199: runtime soap binding error using jar-with-dependencies build

zumi000, 2019-09-17:
#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [x ] I checked the documentation and wiki, but found no answer/solution
- [x ] I am running the latest version and the issue still occurs
- [x ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.3.2
Operating system  : Ubuntu 18.04.3 LTS
JDK               : 11.0.4
Database          : 5.7.27
```

#### Expected Behavior

can run as jar with build in all dependecies, too

#### Actual Behavior

can't

#### Steps to Reproduce the Problem

if i make a jar, with all dependencies, i get a runtime error:
org.apache.cxf.service.factory.ServiceConstructionException: Could not resolve a binding for http://www.w3.org/2003/05/soap/bindings/HTTP/
full log: https://pastebin.com/QNP9i682

with the original maven project i used e.g. the maven assembly plugin, but i i made a gladle version of this full project, but it starts with the same error.

i don't have any more ideas, why i get this runtime error. do you have any ideas or suggestions?

#### Additional context

i have this error only with build-in-dependecies jar. if i use the original maven build script, or my new gradle build script without dependencies, they work fine.

goekay, 2019-09-17:
you didn't really specify how you make a jar with all dependencies. you basically have to copy the contents (read: classes, etc.) of all dependencies (read: jars) into one jar, which is called a _fat jar_. we were using this before, but changed the behaviour: [the relevant issue](https://github.com/RWTH-i5-IDSG/steve/issues/121), [the commit](https://github.com/RWTH-i5-IDSG/steve/commit/b65caaa2094cd233c59bc40a68fa508741f44431).

you should use the maven plugin (maven-shade-plugin) that does what i described above. if you revert the commit in your local repository, or at least add the deleted plugin with its configuration back, you should be able to build a fat jar with the command `mvn package`.

zumi000, 2019-09-25:
i fixed it with using the maven-shade-plugin. thank you for helping me.

closed 2019-09-25 by zumi000