steve-community/steve #202: Make registration status of BootNotification configurable

lategoodbye, 2019-09-19:
#### Checklist

- [X] I checked other issues already and found no answer
- [X] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

I'm using SteVe 3.3.0 to test our OCPP 1.6J client implementation, so it would be nice to test all variants of a registration status in BootNotification.Confirm for a longer time period (OCPP 1.6 = Accepted, Pending, Rejected).

#### Describe the solution you'd like

I'm thinking of a single selectbox or radios in the OCPP settings. Any change should only affect to new BootNotification.Requests. Already running OCPP connections shouldn't be not affected.

goekay, 2019-09-20:
hi, i can see the value of this change. i have some remarks about the current implementation for you and others to give feedback on how to change the behaviour:
- Pending status is currently not implemented.
- Accepted status is sent back if the charging station is registered (i.e. it is present in database)
- Steve does not accept websocket/json connection attempts, if the station is not registered. therefore, a Rejected status is never sent back, if the websocket connection can be established.

lategoodbye, 2019-09-20:
The following decision table comes to my mind:

Central system setting | Station | BootNotification Registration status |
--- | --- | ---
Accepting | registered | Accepted
Accepting | not registered | Rejected
Pending | registered | Pending
Pending | not registered | Rejected
Rejecting | registered | Rejected
Rejecting | not registered | Rejected

goekay, 2019-09-20:
hmm, thanks. so... some takeaways from your table:
- if the station is not registered, always return Rejected (central system setting does not matter).
- otherwise, return the value of central system setting.

i think this decision table is simpler :)

goekay, 2019-09-20:
additional notes for myself while implementing this feature:
- i plan to continue rejecting websocket connection attempts from ocpp-j stations, if they are not registered.
- Pending status is introduced in ocpp 1.6: the selection dropdown values must reflect the station's ocpp version.
- from ocpp 1.6 spec: "While in pending state, the following Central System initiated messages are not allowed: RemoteStartTransaction.req and RemoteStopTransaction.req"

goekay, 2019-10-16:
@lategoodbye i implemented the feature as ilustrated above. can you please test the branch and give feedback, such that i merge the changes into master afterwards? thanks.

lategoodbye, 2019-10-17:
Great. Unfortunately i'm very busy this week. So it might take it until next week.

lategoodbye, 2019-10-25:
Sorry, for the delay. I checkout d6eeb4f0eb9663c78e9b4f3fe96d910e6a454119 from branch issue202 and run steve on my Ubuntu 16.04.

If i change "Registration status" to "Pending" or "Rejected", the charge point is getting a HTTP 500 error trying to opening the websocket with the following content:

```
GET /steve/websocket/CentralSystemService/tarragon_leipzig HTTP/1.1
Connection: Upgrade
Host: 192.168.178.88:8080
Sec-WebSocket-Key: 9UGdtWiZeeIc85/SHBc2EA==
Sec-WebSocket-Protocol: ocpp1.6
Sec-WebSocket-Version: 13
Upgrade: websocket
User-Agent: WebSocket++/0.8.1

HTTP/1.1 500 Server Error
Cache-Control: must-revalidate,no-cache,no-store
Content-Type: text/html;charset=iso-8859-1
Content-Length: 8303
Connection: close

<html>
<head>
<meta http-equiv="Content-Type" content="text/html;charset=utf-8"/>
<title>Error 500 Server Error</title>
</head>
<body><h2>HTTP ERROR 500</h2>
<p>Problem accessing /steve/websocket/CentralSystemService/tarragon_leipzig. Reason:
<pre>    Server Error</pre></p><h3>Caused by:</h3><pre>org.springframework.web.util.NestedServletException: Request processing failed; nested exception is org.springframework.web.socket.server.HandshakeFailureException: ChargeBoxId &apos;tarragon_leipzig&apos; is not registered
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1014)
	at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:898)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:876)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:542)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:146)
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:257)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1711)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:255)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1347)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:203)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:480)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1678)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:201)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1249)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144)
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.handle(GzipHandler.java:753)
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:61)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
	at org.eclipse.jetty.server.Server.handle(Server.java:505)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:370)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:267)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126)
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:781)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:917)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: org.springframework.web.socket.server.HandshakeFailureException: ChargeBoxId &apos;tarragon_leipzig&apos; is not registered
	at de.rwth.idsg.steve.ocpp.ws.OcppWebSocketUpgrader.upgrade(OcppWebSocketUpgrader.java:74)
	at org.springframework.web.socket.server.support.AbstractHandshakeHandler.doHandshake(AbstractHandshakeHandler.java:290)
	at org.springframework.web.socket.server.support.WebSocketHttpRequestHandler.handleRequest(WebSocketHttpRequestHandler.java:168)
	at org.springframework.web.servlet.mvc.HttpRequestHandlerAdapter.handle(HttpRequestHandlerAdapter.java:53)
[... 31 lines cut ...]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144)
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.handle(GzipHandler.java:753)
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:61)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
	at org.eclipse.jetty.server.Server.handle(Server.java:505)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:370)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:267)
[... cut ...]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:781)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:917)
	at java.base/java.lang.Thread.run(Thread.java:834)
</pre>

</body>
</html>
```

I would expect the websocket opens and the BootNotification gets a "Pending" or "Rejected" status. If i set the Registration status to "Accepted" everything works as expected.

goekay, 2019-10-25:
hmm. ok. by design, i wanted to allow connections when Accepted. i can relax the conditions.

lategoodbye, 2019-10-25:
I understand this saves a lot resources. The problem with the behavior (beside it isn't conform to the spec) is that i want to test RegistrationStatus = "Pending" or "Rejected" in the BootNotification.Conf. But so i'll never reach this point, since the websocket is not opened.

Is it possible to make this configurable with a property (In case you are concerned about denial of service):
open.websocket.unknown.stations

goekay, 2019-10-25:
_please see my edit. i am willing to change the impl to allow "Pending". the following applies to the decision about "Rejected"._

it's not about saving resources. the discussion is about finding a tradeoff between being too loose or restrictive in order to prevent some stations accessing the central system and misbehaving.

> ... beside it isn't conform to the spec ...

the question that we try to answer is whether we should treat _the station not in database_  and _the station in database, but with status "Rejected"_ the same or not. unfortunately, this level of granularity is not expressed in spec.

> Is it possible to make this configurable with a property ...

i would prefer not to, if we can find a solution. i'd rather keep the properties as few as possible.

lategoodbye, 2019-10-25:
How about this?

Central system setting | Station | Websocket | BootNotification Registration status
--- | --- | --- | ---
Accepting | registered | Open | Accepted
Accepting | not registered | Close | -
Pending | registered | Open | Pending
Pending | not registered | Close | -
Rejecting | registered | Open | Rejected
Rejecting | not registered | Close | -

goekay, 2019-10-25:
i try to simplify the table, with the addition of italic text (soap case if not registered):

* if not registered (not in db)
  * ws: do not allow websocket connections
  * _soap: return registration status "Rejected"_
* if registered (in db):
  * ws: allow all websocket connections (regardless of registration status)
  * ws and soap: return registration status from db in BootNotification

i think we can agree on this. okay for you?

lategoodbye, 2019-10-25:
Yes, fine.

goekay, 2019-10-25:
do you plan to test the latest commit? otherwise, i am merging the branch and will be closing this issue.

closed 2019-10-25 by goekay

lategoodbye, 2019-10-26:
I'll test the new version if i have access to our charge controller.

lategoodbye, 2019-11-07:
FWIW the new feature works fine in SteVe 3.4.0.

warukunaiyone, 2024-08-20:
Hello, I’d like to ask which versions currently support the Pending status?