steve-community/steve #209: Billing integration

scribas, 2019-10-02:
I would like to find out how other people integrate SteVe with a billing engine? Are there any standard ways that this is implemented? Any specific rating and invoicing solutions that anyone can share?

csamsel, 2019-10-02:
Via the transaction data export

goekay, 2019-11-04:
since there is no communaction happening, here are my 2 cents :)

there are multiple paid complete ocpp solutions that include billing and payment (please google). i am unaware of a solution that only supports a subset of the functionality (i.e. only billing).

> Via the transaction data export
 
yes, but transaction data export is only the first step in the journey of billing. you can use transaction data to map the usage of a station to a monatery value. the missing part in steve is this mapping logic, which @scribas is asking.

> Are there any standard ways that this is implemented?

in transaction data export, there are two standard properties for each transaction that can be used for this purpose:
- the duration that a customer occupied the station for
- the energy (in Wh) that a customer consumed

how you use these properties heavily depends on your business and tariff model, etc.: you can use one or both of them (combined) in order to calculate the price. of course the unit price for duration and energy can be a fixed or dynamic one. you can have multiple tariff types for your customers, you can apply surge pricing, you can leverage smart charging as an incentive for cheaper charging at night rather than during the day, and so on. the possibilities are virtually endless.

i even have a basic implementation for price calculation laying around, where tariffs are defined in java classes. a tariff management is missing (e.g. defining multiple tariffs, their validity time periods, assignment of tariffs to users). therefore, only one tariff can be used for all transactions, for which the price is to be calculated. since the impl is incomplete and java knowhow is necessary to do something useful with it, i was indecisive to commit it in the repo. but i can do it, if the interest is there and other devs want to continue developing it.

scribas, 2019-11-05:
thanks @goekay. i am actually considering fetching the transactional data directly out of the mysql tables and then write some python scripts to do the rating and invoicing. but will have a look for those paid solutions.

goekay, 2019-11-06:
i consider this issue resolved and am closing it.

closed 2019-11-06 by goekay