steve-community/steve #210: An illegal reflective access operation has occurred (Using JDK 11 - Oracle latest version)

MikeK19, 2019-10-11:
How can I make the program runs:
Jave -jar target/steve.jar

And fix the below issues with warnings.

#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [x ] I am running the latest version and the issue still occurs
- [ x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.2.2 (latest)
Operating system  : Windows 10 - 64-bits
JDK               :  JDK 11 (Oracle)
Database          : HeidiSQL (MariaDB 10.4)
```

#### Expected Behavior

Run steve.jar 

Jave -jar target/steve.jar

#### Actual Behavior
Starting....WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.jooq.tools.reflect.Reflect (file:/C:/Steve-3.3.2/target/libs/jooq-3.11.11.jar) to constructor java.lang.invoke.MethodHandles$Lookup(java.lang.Class)
WARNING: Please consider reporting this to the maintainers of org.jooq.tools.reflect.Reflect
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
.......... FAILED!
Please refer to the log file for details
...

#### Steps to Reproduce the Problem
  1. ...
  2. ...
  3. ...

#### Additional context
...

goekay, 2019-10-11:
first of all, the reason why it fails has nothing to do with the warnings. please search in issues to find more info about the warnings. they are not fatal.

i cannot help you with your problem, because as you posted: `Please refer to the log file for details`. Look at the log file for any cues and exceptions.

MikeK19, 2019-10-11:
The mysql server time was not set to UTC. Once I changed the setting and restarted the server, it worked fine. Now I can run the steve.jar file.
Thank you.

closed 2019-10-11 by MikeK19