steve-community/steve #211: How Can the Application Connect to ChargeBox with Encrypted ID?

MikeK19, 2019-10-11:
Hi,

I would like to use the application to test OCPP compliance for a vendor chargers. How can I set the application to connect to the chargers which I don't know their ChargeBox ID (encrypted). The chargers are in different network domain. 

Thank you

goekay, 2019-10-11:
> How can I set the application to connect to the chargers which I don't know their ChargeBox ID (encrypted).

i think you have a misunderstanding how steve works with stations. you configure the stations to connect to steve, and not the other way around. if the stations are not registered in database, you should see exceptions in the log file and can find out the chargebox id there. as an alternative, you can change [this setting](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/prod/main.properties#L51) to accept all communication attemps from unknown stations.

MikeK19, 2019-10-11:
So the charger should be configured to communicate to SteVe.  Did you have a test case that Steve had been tested with ABB chargers? I didn't find ABB is the compatible devices list.

goekay, 2019-10-11:
no. then please test and update the list ;)

closed 2019-10-12 by goekay