steve-community/steve #212: SignedData transfer with a Keba x-serie M&E

jNo89, 2019-10-14:
I have tried out the signed data (SD) transfer with a Keba x-serie M&E and the SteVe backend to verify these data with the 'Transparenzsoftware'. The measured values (start and stop value) were transferred into the database, but I couldn't find the signed measured values in any database. Are these values with the SD (see below) not collected and written into a database?

**Transaction table from SteVe:**
transaction_pk | charge_box_id | connector_id | id_tag | start_timestamp | start_value | stop_timestamp | stop_value | stop_reason
-- | -- | -- | -- | -- | -- | -- | -- | --
40 | EnQS | 1 | enqsTest | 2019-10-11T07:44:32.895Z | **173** | 2019-10-11T07:48:30.070Z | **498** | EVDisconnected

**From the Logging of the ChargingStation:**
2019-10-11 07:48:30,743 [DispatcherThread] DEBUG (WebSocketClient:311 sendRequest) - Sending [2,"65958799-c6d8-43cc-bca2-60992dd7e309","StopTransaction",{"transactionId":40,"idTag":"enqsTest","timestamp":"2019-10-11T07:48:30.070Z","meterStop":498,"reason":"EVDisconnected","transactionData":[{"timestamp":"2019-10-11T07:44:32.895Z","**sampledValue**":[{"value":"173.8","context":"Transaction.Begin","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"}]},{"timestamp":"2019-10-11T07:48:30.070Z","**sampledValue**":[{"value":"498.7","context":"Transaction.End","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"}]},{"timestamp":"2019-10-11T07:44:32.895Z","**sampledValue**":[{"value":"**OCMF|{\"FV\":\"1.0\",\"GI\":\"KEBA_KCP30\",\"GS\":\"0\",\"GV\":\"2.8.6\",\"PG\":\"T7\",\"IS\":false,\"IL\":\"NONE\",\"IF\":[\"RFID_NONE\",\"OCPP_NONE\",\"ISO15118_NONE\",\"PLMN_NONE\"],\"IT\":\"NONE\",\"ID\":\"\",\"RD\":[{\"TM\":\"2019-10-11T07:44:31,000+0000 I\",\"TX\":\"B\",\"EF\":\"\",\"ST\":\"G\",\"RV\":0.1738,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"},{\"TM\":\"2019-10-11T07:48:28,000+0000 I\",\"TX\":\"E\",\"EF\":\"\",\"ST\":\"G\",\"RV\":0.4987,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"}]}|{\"SD\":\"3045022100C80054B6DA58FAA1FE8665144DF330CF4E92C010B2C4497D24A3342CD82FAF63022046A51B4D44C23A5C255C998BE3545E57B5EEA70CE288B1C8184391E43100426A\"}"**,"context":"Transaction.Begin","format":"SignedData","measurand":"Energy.Active.Import.Register"}]},{"timestamp":"2019-10-11T07:48:30.070Z","sampledValue":[{"value":"OCMF|{\"FV\":\"1.0\",\"GI\":\"KEBA_KCP30\",\"GS\":\"0\",\"GV\":\"2.8.6\",\"PG\":\"T7\",\"IS\":false,\"IL\":\"NONE\",\"IF\":[\"RFID_NONE\",\"OCPP_NONE\",\"ISO15118_NONE\",\"PLMN_NONE\"],\"IT\":\"NONE\",\"ID\":\"\",\"RD\":[{\"TM\":\"2019-10-11T07:44:31,000+0000 I\",\"TX\":\"B\",\"EF\":\"\",\"ST\":\"G\",\"RV\":**0.1738**,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"},{\"TM\":\"2019-10-11T07:48:28,000+0000 I\",\"TX\":\"E\",\"EF\":\"\",\"ST\":\"G\",\"RV\":**0.4987**,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"}]}|{\"SD\":\"3045022100C80054B6DA58FAA1FE8665144DF330CF4E92C010B2C4497D24A3342CD82FAF63022046A51B4D44C23A5C255C998BE3545E57B5EEA70CE288B1C8184391E43100426A\"}","context":"Transaction.End","format":"SignedData","measurand":"Energy.Active.Import.Register"}]}]}]

**From the Logging fo the ChargingStation:**
mva_record {
  record_type: MEASUREMENT
  record_id: 7
  data: "{\"FV\":\"1.0\",\"GI\":\"KEBA_KCP30\",\"GS\":\"0\",\"GV\":\"2.8.6\",\"PG\":\"T7\",\"IS\":false,\"IL\":\"NONE\",\"IF\":[\"RFID_NONE\",\"OCPP_NONE\",\"ISO15118_NONE\",\"PLMN_NONE\"],\"IT\":\"NONE\",\"ID\":\"\",\"RD\":[{\"TM\":\"2019-10-11T07:44:31,000+0000 I\",\"TX\":\"B\",\"EF\":\"\",\"ST\":\"G\",\"RV\":**0.1738**,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"},{\"TM\":\"2019-10-11T07:48:28,000+0000 I\",\"TX\":\"E\",\"EF\":\"\",\"ST\":\"G\",\"RV\":**0.4987**,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"}]}"
  signature: "{\"SD\":\"3045022100C80054B6DA58FAA1FE8665144DF330CF4E92C010B2C4497D24A3342CD82FAF63022046A51B4D44C23A5C255C998BE3545E57B5EEA70CE288B1C8184391E43100426A\"}"
}

Many thanks in advance!

goekay, 2019-10-14:
it should be written to database, but steve cannot dive deeper into the data structure of `"value":{...}` since this is manufacturer-specific. therefore, you should see the whole field starting with `"OCMF|...` there and not only SD.

moreover, we do not differentiate between the various `sampledValue` possibilities and just dump them to database without any logic. this means, if you can see measured start and stop values in `meter_value` table, you should see the other entries as well. make sure from steve's log file that there are no exceptions and problems while processing the message.

jNo89, 2019-10-14:
thanks you for the quick feedback! The complete field 'OCMF|...' would be perfect and exactly what I need. 
Do you mean with 'meter_value' the table transactions? Or does SteVe V3.2.0 not inclued the table meter_value?

goekay, 2019-10-14:
> Do you mean with 'meter_value' the table transactions? Or does SteVe V3.2.0 not inclued the table meter_value?

the following tables are relevant to this discussion and your problem:
- `transaction_start` and `transaction_stop`: these store all the transaction-related basic info like station connector, id tag, timestamp and meter value for start and stop events, respectively.
- `connector_meter_value`: as you can see from your message there are additional meta-info that a station might want to send. this can be related to a transaction, or not. this table stores this kind of meta-info or intermediate meter values between transaction start and stop events.

you should look for the compelete field `OCMF|...` in the  `connector_meter_value` table.

jNo89, 2019-10-14:
thats all I can found related to the transaction_pk = 40 
[image]

goekay, 2019-10-14:
hmm... that should not be the case. can you please look at the log file whether there are any exceptions and let me know about them?

goekay, 2019-10-14:
i have a feeling what the cause of your problem might be: the value field has a data type of `varchar(255)`, which means that a textual value up to 255 characters is allowed. the value that you are trying to insert exceeds that upper bound.

i am planning to upgrade this data type. in case of SignedData, the value field in the message is opaque and can be of any arbitrary length as the manufacturer wishes. which makes the decision harder.

i am thinking of the data type `TEXT`, which allows up to 65,535 characters. i also contemplate data types like `MEDIUMTEXT` or `LONGTEXT` and therefore virtually remove the upper bound, but they might be overkill for most usecases where we store Raw values for units like Watts, Ampere, Celsius, etc.

i am waiting for your feedback (exceptions in log file) to confirm my suspicion. on the other hand, any other ideas are welcome.

jNo89, 2019-10-15:
that's a good hint. I changed the dataType from the field value to 'text' and scheduled a test for tomorrow. I will report how it worked. 

it would also be interesting to integrate the 'context' from the sampledValue into the table in order to distinguish the start and end of a load process more easily. 

Sending [2,"65958799-c6d8-43cc-bca2-60992dd7e309","StopTransaction",{"transactionId":40,"idTag":"enqsTest","timestamp":"2019-10-11T07:48:30.070Z","meterStop":498,"reason":"EVDisconnected","transactionData":[{"timestamp":"2019-10-11T07:44:32.895Z","sampledValue":[{"value":"173.8",**"context":"Transaction.Begin"**,"format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"}]},{"timestamp":"2019-10-11T07:48:30.070Z","sampledValue":[{"value":"498.7",**"context":"Transaction.End"**,"format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"}]},{"timestamp":"2019-10-11T07:44:32.895Z","sampledValue":[{"value":"OCMF|{\"FV\":\"1.0\",\"GI\":\"KEBA_KCP30\",\"GS\":\"0\",\"GV\":\"2.8.6\",\"PG\":\"T7\",\"IS\":false,\"IL\":\"NONE\",\"IF\":[\"RFID_NONE\",\"OCPP_NONE\",\"ISO15118_NONE\",\"PLMN_NONE\"],\"IT\":\"NONE\",\"ID\":\"\",\"RD\":[{\"TM\":\"2019-10-11T07:44:31,000+0000 I\",\"TX\":\"B\",\"EF\":\"\",\"ST\":\"G\",\"RV\":0.1738,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"},{\"TM\":\"2019-10-11T07:48:28,000+0000 I\",\"TX\":\"E\",\"EF\":\"\",\"ST\":\"G\",\"RV\":0.4987,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"}]}|{\"SD\":\"3045022100C80054B6DA58FAA1FE8665144DF330CF4E92C010B2C4497D24A3342CD82FAF63022046A51B4D44C23A5C255C998BE3545E57B5EEA70CE288B1C8184391E43100426A\"}",**"context":"Transaction.Begin"**,"format":"SignedData","measurand":"Energy.Active.Import.Register"}]},{"timestamp":"2019-10-11T07:48:30.070Z","sampledValue":[{"value":"OCMF|{\"FV\":\"1.0\",\"GI\":\"KEBA_KCP30\",\"GS\":\"0\",\"GV\":\"2.8.6\",\"PG\":\"T7\",\"IS\":false,\"IL\":\"NONE\",\"IF\":[\"RFID_NONE\",\"OCPP_NONE\",\"ISO15118_NONE\",\"PLMN_NONE\"],\"IT\":\"NONE\",\"ID\":\"\",\"RD\":[{\"TM\":\"2019-10-11T07:44:31,000+0000 I\",\"TX\":\"B\",\"EF\":\"\",\"ST\":\"G\",\"RV\":0.1738,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"},{\"TM\":\"2019-10-11T07:48:28,000+0000 I\",\"TX\":\"E\",\"EF\":\"\",\"ST\":\"G\",\"RV\":0.4987,\"RI\":\"1-b:1.8.0\",\"RU\":\"kWh\"}]}|{\"SD\":\"3045022100C80054B6DA58FAA1FE8665144DF330CF4E92C010B2C4497D24A3342CD82FAF63022046A51B4D44C23A5C255C998BE3545E57B5EEA70CE288B1C8184391E43100426A\"}"**,"context":"Transaction.End"**,"format":"SignedData","measurand":"Energy.Active.Import.Register"}]}]}]

goekay, 2019-10-15:
> it would also be interesting to integrate the 'context' from the sampledValue into the table in order to distinguish the start and end of a load process more easily.

the column `reading_context` is already there for this. basically, we insert _everything_ from `sampledValue` items into the table. there should not be any information loss.

jNo89, 2019-10-16:
Today's test was successful. With the change of the dataTypes suggested by you, the signed measured values were written into the database! In addition, the signed measured values could be successfully verified via the 'Transparenzsoftware' and the reading_context was also wirtten into the database. 

[image]

So in generell SteVe is fully working with the Keba x-serie M&E.
Thanks for your work and your support!

goekay, 2019-10-16:
this is good to hear. in the near future, i will add another [database migration script](https://github.com/RWTH-i5-IDSG/steve/tree/master/src/main/resources/db/migration) to change this field's data type to `TEXT` for the benefit of everyone. please keep the following in mind: since you manually modified the database schema, you might encounter problems when updating steve, after i add the aforementioned migration.

commit f3b065b4 (2019-10-24): upgrade db column data type to TEXT (closes #212)
closed 2019-10-24 by goekay