steve-community/steve #217: "Max. Active Transaction Count" condition not being evaluated

anirudh-ramesh, 2019-10-22:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.3.2 (commit 40c5d00)
Operating system  : Ubuntu 18.04.3 LTS
JDK               : openjdk 11.0.2 2019-01-15
Database          : 1.0.0 (commit 40c5d00)
```

#### Expected Behavior

When 'max. active transaction count' is set at 1 and a user having an authorized tag attempts to start a transaction when another transaction corresponding to the same tag is active at another charging point, the first transaction should continue and the second transaction should not be approved.

#### Actual Behavior

The second transaction is approved and the first transaction continues undisturbed.

#### Steps to Reproduce the Problem

  1. Create 2 charge points, 1 user and 1 tag that has 'max. active transaction count' set to 1, and assigned to the same user.
  2. Connect both charging points to SteVe.
  3. Authorize tag at first charging point and start a transaction. Ensure the transaction is active.
  4. Authorize tag at second charging point and start a transaction. View tag properties.

#### Additional context

N/A

goekay, 2019-10-22:
i added a test which verifies the authorization status of the second authorization attempt. the test passes. are you sure about this?

> 4. Authorize tag at second charging point and start a transaction. View tag properties.

hmm maybe there is a confusion about the flow of operations. the charging station should ask to auth the tag and start a transaction only if the authorization status of the tag is accepted. steve always accepts all start and stop transaction events, since they might be from the past which cannot be sent to backend (steve) due to communication errors.

am i right to assume that in your flow the auth response is ignored by the station and transaction is started no matter what?

anirudh-ramesh, 2019-10-22:
Yes, I think the flow of operations is a little different in my case - I am authorizing the tag at two different charging points when neither charging point has an active transaction - therefore, SteVe would accept both authorization requests.

Subsequently, when there's a request to initiate the transaction - like you are saying - SteVe would not evaluate the condition for max active transaction count.

So, I see something the following.
[image]

goekay, 2019-10-22:
> Subsequently, when there's a request to initiate the transaction ...

the interpretation is a little different than that and the nuance is important. StartTransaction message is not a request. it is merely a notification about an already started transaction. the _gatekeeper_ was the AuthRequest.

> ... SteVe would not evaluate the condition for max active transaction count.

what do you mean by that?

let me try to illustrate current behaviour and (hopefully if i understand your point) what should happen from your perspective:

1. auth message to station1. status in response is `Accepted`.
2. auth message to station2. status in response is `Accepted`.
3. transaction start message to station1. response message contains the status `Accepted` in tag info.
4. transaction start message to station2. response message contains the status `Accepted` in tag info. you want this status to be `ConcurrentTx` such that the station can react to it at least after the fact.

anirudh-ramesh, 2019-10-22:
> StartTransaction message is not a request. it is merely a notification about an already started transaction. the gatekeeper was the AuthRequest.

I see - I had earlier understood that "Start Transaction" was a request - thanks for pointing out and correcting me!

> ... SteVe would not evaluate the condition for max active transaction count.

What I mean here is that since authorization of the tag in both sessions had been accepted, SteVe would not evaluate this condition when the 'start transaction' is issued (I had written this upon incorrectly understanding that 'start transaction' is a request and not a notification; since it is actually a notification, evaluating a condition is meaningless).

> 1. auth message to station1. status in response is Accepted.
> 2. auth message to station2. status in response is Accepted.
> 3. transaction start message to station1. response message contains the [...]

This sequence accurately describes what is going on on my end. I now think this could be considered a corner case and not necessarily a bug in SteVe since this behaviour also depends on the charging point and session as not all points or sessions may have a significant time duration between tag authorization 1 and transaction start notification 1 that a tag authorization 2 could happen in the meantime, but it is not completely impossible either, so it would be good to document this corner case at the least.

goekay, 2019-10-22:
active and max active transaction count only affect the authorization status that is set in responses to auth and start/stop transaction requests. other than that, they are not used when processing a start transaction request whether the max count is already reached and this transaction request therefore should be rejected. _the rejection of a transaction request_ is not possible, since the decision is done by the station already (and possibly earlier than the time when backend is informed). therefore, we _always_ write this data into the database as the start of a new transaction.

active transaction count is nothing more than the number of transactions for which we received a start message but not yet a stop message.

anirudh-ramesh, 2019-10-23:
Okay, I see; this makes sense.

Does this incident have to be categorized as a bug in the OCPP client?

How do other (perhaps proprietary) charging station management systems typically ringfence vulnerabilities arising out of corner cases like this e.g. could the tag authorization expire in a minute or two, or would it be necessary to periodically scan the number of active transactions associated with a particular tag and issue alarms or forcibly stop certain transactions?

goekay, 2019-10-23:
ocpp always had a focus on offline and standalone operation of stations to circumvent connectivity and network problems. this is realized by uploading the id tag list to the station with SendLocalList. afterwards, the station has a complete knowledge about the id tags. keeping the stations up-to-date with changes is an operational problem. SendLocalList can be repeated daily, weekly, monthly, when something changes etc. this should answer your question about authorization expire, too: the station knows about this stuff in advance.

as you can imagine, in such offline and standalone operations, concurrency control of transactions is not possible, since the central entity (backend) keeps the global state, which is unreachable. therefore, i would argue that such a strict concurrency control is not a design focus of ocpp. nevertheless, in your case, steve can set the id tag status to `ConcurrentTx` of the second start transaction response, to which the station should react by immediately stopping this started transaction. i guess, _better late than never_ is the motto here. 

but there is a caveat: what if the first transaction was already stopped, but the station became offline and could not send the StopTransaction message to backend and therefore this id tag remains falsely in status `ConcurrentTx` and the second transaction was stopped undeservedly?

csamsel, 2019-10-23:
> How do other (perhaps proprietary) charging station management systems typically ringfence vulnerabilities arising out of corner cases like this

Using the the same charging card twice in different places usually dont work for obvious reasons anyways.

goekay, 2019-10-23:
@csamsel yeah but i can imagine a parking lot with many nearby stations. this is a valid concern.

anirudh-ramesh, 2019-10-23:
@goekay,

> this should answer your question about authorization expire, too

Yes, it answers my question indeed, thanks!

> steve can set the id tag status to ConcurrentTx of the second start transaction response, to which the station should react by immediately stopping this started transaction

I had a chance to check out the OCPP 1.6 client simulator earlier today with SteVe and understood this behaviour better - I think the client software should follow best practices and be appropriately designed.

@csamsel,

> Using the the same charging card twice in different places usually dont work for obvious reasons anyways

> yeah but i can imagine a parking lot with many nearby stations. this is a valid concern.

Agreed, and since this would become a major vulnerability in cases of shoddily designed OCPP client software, it would be necessary to define an SLA that the charging station management system operator isn't accountable in a scenario like this as it could likely be a business case for a penalty or any other contractual obligation.

Thanks, @goekay, for this discussion and the diagnosis - I think this issue is well documented, and perhaps resolved too, at least in part.

closed 2019-10-23 by goekay