steve-community/steve #218: Set Charging Profile

jNo89, 2019-10-23:
We tried a few tests on the smart chagring via SteVe and found that the charging profiles were not executed as specified from us. During the analysis I noticed that the unit that arrives at the charging station is not Ampere but milli Ampere. Could it therefore be that the unit in the charging profile is not A but mA? 
I also noticed when I set the Unit to W instead of A that the charging station received a calculated Amp value which I did not expected. Is the conversion based on a DC charging?

**Log from the charging station:**
2019-10-22 08:24:24,362 [ tyrus-jdk-client-84] DEBUG (WebSocketClient$2:446 onMessage) - Received message [2,"f7916699-8f07-4e37-97cb-ad8d7492ed32","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":0,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","recurrencyKind":"Daily","validFrom":"2019-10-22T08:25:00.000Z","validTo":"2019-10-22T08:36:00.000Z","chargingSchedule":{"startSchedule":"2019-10-22T08:26:00.000Z","**chargingRateUnit":"A"**,"chargingSchedulePeriod":[{"startPeriod":0,**"limit":1**,"numberPhases":3}]}}}]
2019-10-22 08:24:24,545 [EventLoopThread-1571730183499]  INFO (LogRuleName:497 logRuleName) - RULE [NO_SERIAL_RULE] ---IF SetChargingProfileCPMRequest THEN call actor. Send ACCEPTED SetChargingProfileCPMReponse to host.---
2019-10-22 08:24:24,698 [ tyrus-jdk-client-84] DEBUG (WebSocketClient:386 sendResponse) - Answering [3,"f7916699-8f07-4e37-97cb-ad8d7492ed32",{"status":"Accepted"}]
2019-10-22 08:24:24,774 [Actor-SetChargingProfileActorImpl-0]  INFO (MaxScheduleServiceImpl:60 scheduleMaxScheduleTask) - task scheduled to execute after 35469 milliseconds for max schedule MaxSchedule {id = [2] maxScheduleTimeReceived = [Tue Oct 22 08:24:24 UTC 2019] maxScheduleValidFrom = [Tue Oct 22 08:25:00 UTC 2019] maxScheduleValidTo = [Tue Oct 22 08:36:00 UTC 2019] maxScheduleStart = [Tue Oct 22 08:26:00 UTC 2019] maxScheduleSetID = [10] maxScheduleID = [10] maxScheduleKind = [ONE_TIME] powerProfileId = [null] items = [
 MaxScheduleItem {maxScheduleItemStart = [0 s] maxScheduleItemStop = [660 s] **maxScheduleMaxCurrent = [1 mA]** **maxScheduleMaxPower = [1 W]** maxScheduleNumberPhases = [3] }]}

**ChargingProfil in SteVe:**
[image]

goekay, 2019-10-23:
> During the analysis I noticed that the unit that arrives at the charging station is not Ampere but milli Ampere.

the first part of the log describes the arriving message at the station: 

> 2019-10-22 08:24:24,362 [ tyrus-jdk-client-84] DEBUG (WebSocketClient$2:446 onMessage) - Received message [...]

the unit is Ampere and the values match the values from screenshot. the following next part from the log is the internal logic of the station and i can imagine that something goes wrong within mapping logic of the station:

> 2019-10-22 08:24:24,774 [Actor-SetChargingProfileActorImpl-0] INFO (MaxScheduleServiceImpl:60 scheduleMaxScheduleTask) - task scheduled to execute after 35469 milliseconds for max schedule [...]
MaxScheduleItem {maxScheduleItemStart = [0 s] maxScheduleItemStop = [660 s] maxScheduleMaxCurrent = [1 mA] maxScheduleMaxPower = [1 W] maxScheduleNumberPhases = [3] }]}

jNo89, 2019-10-23:
thank you very much for the feedback. I will use it to approach Keba and report about the problem.

goekay, 2019-10-23:
ok. you can open the issue again, if it turns out to be a steve problem.

p.s.: please use issue templates.

closed 2019-10-23 by goekay