steve-community/steve #242: RFC: Move ocpp-jaxb repository from internal nexus to a public repository

csamsel, 2019-11-25:
Long term it's not feasible to host the https://github.com/RWTH-i5-IDSG/ocpp-jaxb artifacts on idsg nexus' as it might disappear for what ever reason. 

I suggest to support building SteVe easily in the future we move `ocpp-jaxb` somewhere else. 
If you dont agree please speak up.

First lets start by collecting options and the have a look at advantages and disadvantages

Options (no particular order)

1. GitHub Packages https://github.com/features/packages
2. Maven Central https://search.maven.org
3. ?

goekay, 2019-11-25:
i don't see an urgent need to change it proactively. especially, when the source is open and the repo is on github (in worst case, a user can pull the repo, build and install locally). nexus@idsg has been serving us well. we had only one minor problem with it so far (#130). if something happens in future, we can react and _then_ change it (_if it ain't broke, don't fix it)_.

csamsel, 2019-11-25:
Definitely not urgent, but i rather have it somewhere reliable before it breaks. Also pushing the artifact should not be big deal, is it?

Any opinion on GitHub Packages vs. Maven Central?

goekay, 2020-01-27:
i would prefer maven central, since it is _the_ standard repository for java libraries. it would also be good to not completely depend on github. i can look into it.

goekay, 2020-02-01:
as a first step and because of ease of its use, i thought of using jitpack.io, which is a service that pulls a github repo, builds it and hosts on their own artifact repository.

the build of the release version was successful, which can be seen from the logs here: https://jitpack.io/#RWTH-i5-IDSG/ocpp-jaxb

@csamsel what do you think of that? i didn't want to bother with the ceremony/bureaucracy that was necessary for maven central.

csamsel, 2020-02-01:
Hi, sounds good to me. I thought we could do something similar with github packages.

Maybe for an transition time we can have both the old nexus as well as jitpack.io in the steve pom?