steve-community/steve #245: Is it acceptable to use a HeartBeat instead of a Ping-Pong to test a WS connection?

DeltaVetal26, 2019-12-07:
Hi,

I`m using Steve 3.4.1 for OCPP 1.6J.

After reviewing the OCPP1.6 documentation, I learned that OCPP can be used with both HearBeat and Ping-Pong.

HeartBeat is used to synchronize the current time for a charging point. In Steve’s settings, the HeartBeat interval is 4 hours.

Tell me, is this considered normal if I use HeartBeat to test the connection and synchronize the time every 1 minute?

Will this be spam for a WebSocket server?

I haven't figured out Ping-Pong yet, so one way to maintain a WebSocket connection is to send HeartBeat frequently.

Regards,
Alex

goekay, 2019-12-07:
> In Steve’s settings, the HeartBeat interval is 4 hours.

this is the default value, yes, but you can change it.

> Tell me, is this considered normal if I use HeartBeat to test the connection and synchronize the time every 1 minute?

sure, you can do that. but 1 minute interval is probably an overkill. depending on your network configuration and server implementation, a connection should be kept open longer than that when idle. on the other hand, ping-pong will have a smaller overhead for server and client when used frequently and multiple stations, since ping-pong is handled on the lower levels while heartbeat messages reach higher levels of implementation.

DeltaVetal26, 2019-12-08:
@goekay 

Thank you for the answer!

Then I will study the implementation of Ping-Pong.

closed 2019-12-08 by DeltaVetal26