steve-community/steve #249: Stopping a zombie transaction results in java.lang.NumberFormatException

lategoodbye, 2019-12-11:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.3.0
Operating system  : Linux
JDK               : 11
Database          : 0.9.9
```

#### Expected Behavior

I want stop a zombie transaction in the transaction overview and i expected after pushing the red stop button the transaction to be stopped.

#### Actual Behavior

Today i want to stop a zombie transaction in SteVe, but i always get a java.lang.NumberFormatException for this specific transaction (latest one, id 2154). This is strange because the older doesn't trigger this exception.

#### Steps to Reproduce the Problem

Unfortunately i don't know how to get the transaction into this state. My SteVe instance is still in this state, so i'm able to analyse in case you need more information.

#### Additional context

Here is the stacktrace:

```
[ERROR] 2019-12-11 16:01:13,963 de.rwth.idsg.steve.web.GlobalControllerAdvice - Request: http://x.x.x.x/steve/manager/transactions/stop/2154 raised following exception.
java.lang.NumberFormatException: empty String
	at jdk.internal.math.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:1842) ~[?:?]
	at jdk.internal.math.FloatingDecimal.parseDouble(FloatingDecimal.java:110) ~[?:?]
	at java.lang.Double.parseDouble(Double.java:543) ~[?:?]
	at de.rwth.idsg.steve.service.TransactionStopService.floatingStringToIntString(TransactionStopService.java:159) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.TransactionStopService.findNeededValues(TransactionStopService.java:90) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.TransactionStopService.stop(TransactionStopService.java:63) ~[steve.jar:?]
	at de.rwth.idsg.steve.web.controller.TransactionsReservationsController.stopTransaction(TransactionsReservationsController.java:87) ~[steve.jar:?]
	at jdk.internal.reflect.GeneratedMethodAccessor184.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:189) ~[spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:138) ~[spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:102) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:892) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:797) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1038) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:942) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1005) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:908) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:882) ~[spring-webmvc-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:873) ~[jetty-servlet-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1623) ~[jetty-servlet-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:320) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:127) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:91) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:119) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:137) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:111) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:170) ~[spring-security-web-5.1.5.RELEASE.jar:5.1.5.RELEASE]
[... 33 lines cut ...]
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1345) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:203) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:480) ~[jetty-servlet-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1667) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:201) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1247) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.handle(GzipHandler.java:703) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:61) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.Server.handle(Server.java:505) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:370) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:267) ~[jetty-server-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305) ~[jetty-io-9.4.17.v20190418.jar:9.4.17.v20190418]
[... cut ...]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:786) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:743) ~[jetty-util-9.4.17.v20190418.jar:9.4.17.v20190418]
	at java.lang.Thread.run(Thread.java:834) [?:?]
```

goekay, 2019-12-11:
thanks for the report!

the problem occurs at a point where we try to find the last sent meter value and use it as the stop transaction value. i could add a check to bypass null or empty values in this part of the code, but then we would fail to find a reasonable value for stop transaction. therefore we have to dig a little bit deeper when debugging this situation. can you please tell me the last kWh or Wh meter value for this transaction?

lategoodbye, 2019-12-12:
I hope this screenshot is enough:

[image]

goekay, 2019-12-12:
as you can see from the table below in screenshot, there is one meter value 350, that steve should be able to select (even though _Unit_ is not set to Wh or kWh. please report this to the charging station manufacturer. this is a bug!)

the exception in your log file hints at a meter value entry with no value (empty string instead of 350) which is not the case here.

lategoodbye, 2019-12-12:
> as you can see from the table below in screenshot, there is one meter value 350, that steve should be able to select (even though _Unit_ is not set to Wh or kWh. please report this to the charging station manufacturer. this is a bug!)

The unit is optional according to OCPP 1.6. As the measurand is also missing which means Energy.Active.Import.Register, so the unit defaults to Wh.

> the exception in your log file hints at a meter value entry with no value (empty string instead of 350) which is not the case here.

What do you suggest? A database dump?

goekay, 2019-12-12:
yes, please send the database dump to my e-mail address.

goekay, 2019-12-13:
i could not reproduce the problem `java.lang.NumberFormatException: empty String` with the database. the behaviour is as intended. the data shown in screenshot is correct and the transaction can be stopped. since it was a development environment database, i assume incosistent/problematic data as the source of the problem.

i am closing this issue. you can re-open it if you have further evidence.

closed 2019-12-13 by goekay