steve-community/steve #250: statusNotification status update not working

eviontech, 2019-12-13:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : latest
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

status should change to available

#### Actual Behavior
status should change but status remains preparing as initially set..

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...
[image]

[image]

goekay, 2019-12-13:
the timestamp of the status message with "preparing" (5th line in logs) is later than the status with "available". therefore, steve selects the most recent status.

eviontech, 2019-12-13:
[image]

goekay, 2019-12-13:
the time when we receive the messages does not matter. the timestamp of the message counts. there can be thousand messages before `11:08:51`. steve will still pick this one since it is the most recent. probably, the station had a wrong internal clock and the message was queued before applying the current time in response to boot notification.

eviontech, 2019-12-13:
ok understood
thanx

goekay, 2019-12-13:
to further clarify: the time of the log line does not matter. the value of the timestamp property in the message counts.

eviontech, 2019-12-13:
a quick question other than the topic raised

can we change web ui timezone to IST in anyway.... only the display

eviontech, 2019-12-13:
and also can we remove functionalities from our setup ... like we are not implementing parent id concept

goekay, 2019-12-13:
you are targeting multiple topics in one issue. github issues are not a support channel. please be mindful of that. only open issues if you believe that it is a problem with steve. and please use issue templates to rule out duplicates.

> can we change web ui timezone to IST in anyway.... only the display

no, there is an open issue for this feature request: https://github.com/RWTH-i5-IDSG/steve/issues/96

> and also can we remove functionalities from our setup ... like we are not implementing parent id concept

then don't use it. steve supports multiple features to cover various different use cases. if you do not need parent id tags, do not set them in id tags.

closed 2019-12-13 by goekay