steve-community/steve #251: Is it possible to get the latest connector status and idTag of an active transaction from Steve?

DeltaVetal26, 2019-12-13:
Hi,

Example: a charging station has an active session on any of the connectors launched using an RFID card. At some point, a power failure occurs and the station restarts. There is no local storage of RFID tags and the latest status. As a result, charging is suspended, and the station does not know from which card the last charge was made.

Is there an option for such situations with a request for the status of connectors and active transactions from the OCPP server (Steve). I studied the OCPP1.6 documentation, where I did not see such commands, it was only about local storage on the charging station.

Regards,
Alex

goekay, 2019-12-13:
in normal (online) operation cases, the station should inform the server with a StartTransaction. therefore, server should already know the idtag of the active transaction. in offline or standalone operation cases, a local storage must exist in order to authorize charging session attempts.

you can look into the newly introduced TriggerMessage. with this the user of the server can trigger the station to send a message type. StatusNotification is one of the possible types.

hope this helps.

DeltaVetal26, 2019-12-13:
@goekay ,

Thanks for the answer!

Yes, the station can send StartTransaction, where there will be an idTag, which will be displayed in Steve's active transactions. It works. But, if the station suddenly restarts during an active transaction, then all data on the side of the charging station will be lost (if there is no local storage). In this case, it would be nice to send a request to the OCPP server after restoring power, from which you can get information about the last active transaction at the charging station, for example, to automatically resume charging without re-applying RFID.

I will explore the possibilities of TriggerMessage.

goekay, 2019-12-13:
>  But, if the station suddenly restarts during an active transaction, then all data on the side of the charging station will be lost (if there is no local storage). 

do you have a specific, real-world problem or are we talking about hypothetical cases? because if there is no local storage in the charging station _at all_, the negative consequences of this would be far greater than just forgetting the id tag of a transaction. there are various operational situations where persisting data in local storage is necessary.

but to answer your question, no, afaik there is no such ocpp operation to realize what you want.

DeltaVetal26, 2019-12-13:
@goekay ,

I'm creating my OCPP controller, which is connected to an RFID reader and a charging station controller (SmartEVSE). As an OCPP client, I use ESP32, which has the ability to implement storage, but I have not yet moved on to this, as I am not sure of the reliability of data storage.

As I understand it, OCPP 1.6 does not provide such a function.

Then I have two options:
1) Create a normal local storage for transactions.
2) Implement the function of receiving active transactions.

Most likely a more promising option is local storage.

closed 2019-12-15 by DeltaVetal26